riscv-software-src/riscv-isa-sim #546: how is the encoding.h generated

weiqi7777, 2020-09-16:
hi:
    riscv-opcodes tool can generate encoding.h file. 
    and I regenerated the encoding.h file with riscv-opcodes tool(branch master).  but I use the new encoding.h to compile spike , spike failed,  it prompts some  macro don't defined, and miss some instruction.
    I am wonder how the encoding.h file generated in spike?

thanks
lujun

aswaterman, 2020-09-16:
It is generated from riscv-opcodes, but probably from an older version.

weiqi7777, 2020-09-16:
and can we know what's the version of riscv-opcodes to generate the encoding.h  in spike master branch?

thanks
lujun

fvmutlu, 2020-11-21:
I was just dealing with this issue today. It seems that the current version of riscv-opcodes doesn't generate some macros that are used for the current version of Spike. The file I generated was missing quite a few macros, for instance FCVT_Q_H and similar instructions. My workaround was simply using the actual encoding.h from the Spike files, using riscv-opcodes to generate a separate file and just copy what I needed from that into encoding.h. Hope this helps.

chihminchao, 2020-11-23:
riscv-opcodes and riscv-isa-sim are sometimes unsynced and there are some different reasons.

1. The change goes first into spike but the related  part doesn't be included in opcodes
    a.  the previous virtualization support change (has been included)
    b. the zfh (in zfh branch)
    c. removal of quad instruction for spec (PR hasn't been sent)
    d. the b-extension
2. The change is included in riscv-opcodes but spike doesn't has it 
    a. removal of RV128 quad instruction

closed 2021-07-29 by weiqi7777