riscv-software-src/riscv-isa-sim #547: How spike init memory

Phantom1003, 2020-09-16:
I read the codes in fesvr, I found that elf will load in a memif class, when sim class runs.
But I can not find the place, where the code is loaded to memory, I means since the mem is created by following codes in spike.cc
```
mems = make_mems(XXX);
```
How `mems` get the text of elf ?

closed 2020-09-17 by Phantom1003

omerguzelelectronicguy, 2022-05-16:
Hello, @Phantom1003 If you find the answer, could you write it because I need help with a similar issue. I want to locate my programs as starting from 0x40000000 instead of 0x80000000. However, when I prepare a link file that locates the program 0x40000000, it does not allow me to access that region of ram. The error message is 
```
Access exception occurred while loading payload hword.riscv:
Memory address 0x70000290 is invalid
```
Thanks

Phantom1003, 2022-05-16:
spike uses the mem_if interface to load instructions from the elf. For your question, I believe you also need to create a memory at 0x40000000 by using the `-m` option.

omerguzelelectronicguy, 2022-05-16:
In the help message, it says `-m<n>                 Provide <n> MiB of target memory [default 2048]`.
I gave the command as `spike -d -m1024 hword.riscv` However, it again gave me the same error. When I gave as 0x70000000 or anything else less than 0x80000000 it gave me the same error in 0x290 according to the base. So I think I was able to locate the program in that region but it does not allow me to do something. `mret` this is the instruction at 0x290 in the relative address. I am a beginner in spike so excuse me if I missed or misunderstand some clear points.

omerguzelelectronicguy, 2022-05-17:
Exactly, Thank you so much. I wrote it in that format and it is accepted.