riscv-software-src/riscv-isa-sim #551: when v=1, it should report virtual instruction exception instead of illegal instruction exception when executing hfence

tianhu-zeng, 2020-09-24:
According to Spec definition.
"When V=1, a virtual instruction trap (not an illegal instruction trap) is taken for: 
     - attempts to execute a hypervisor instruction (HLV, HLVX, HSV, or HFENCE) or to access an implemented hypervisor CSR or VS CSR;"

But spike reports an illegal instruction exception.

@avpatel, please have a check, thanks.

avpatel, 2020-09-24:
Yes, I am aware of this.

The Hypervisor instruction emulation uses require_novirt() macro which ensures that we generate virtual instruction trap when Hypervisor instruction is executed with V=1

Regards,
Anup

aswaterman, 2020-09-25:
I think the issue is that `require_novirt` comes after `require_privilege`, giving higher priority to the illegal instruction trap.  I will fix this one.

closed 2020-09-25 by aswaterman

tianhu-zeng, 2020-09-25:
@avpatel @aswaterman hfence has the same issue

aswaterman, 2020-09-25:
Fixed.

commit 77024fa1 (2020-09-25): Fix priority of virtual vs. illegal instruction exceptions for HLV/HSV

Resolves #551