riscv-software-src/riscv-isa-sim #553: A question about [m/s/vs]status.[sd,fs]

tianhu-zeng, 2020-09-27:
Hi, 
    When I verify our hypersior extentation by Spike. There are confusions about the FS/SD fields and vsstatus. 
    1. From Spec description, the SD bit is a read-only bit, but it's writable in Spike
    2. In Spike, there is not the SD bit updating logic. Does it update same as the mstatus.SD?
    3. In VS mode, access "sstatus" actually access "vsstatus" instead. It seems that Spike access sstatus directly.
    4. In Spec, "In systems that do not implement S-mode and do not have a floating-point unit, the FS field is
hardwired to zero". I'm curious, in system with M/S/U mode and without F-extentation, what is FS's function? Why don't need to check F-extentation supproted when update [M/S]STATUS.FS. But vsstatus will check F-extentation supproted.

[image]
[image]
[image]
[image]

aswaterman, 2020-09-27:
I'm just writing to note that the SD logic appears correct for mstatus.  I'll leave the rest to @avpatel.

tianhu-zeng, 2020-09-27:
Another issue for hypervisor. when mcounteren = 1 and hcounteren = 0,  access user Counter/Timers csrs should report virtual instruction exception in VS/VU. It seems that the logic is not complete in Spike.

aswaterman, 2020-09-27:
@tianhu-zeng Please don’t create additional confusion by reporting multiple issues together. File a second issue if it isn’t related to the first.

tianhu-zeng, 2020-09-27:
ok, I'll pay more attention to them next time.

avpatel, 2020-09-27:
For 1) and 2), I think @aswaterman already answered your question.

For 3), Spike uses CSR swapping approach for emulating MSTATUS, SSTATUS and VSSTATUS. This is done due to complicated nature of all xSTATUS CSRs. When V=0, the S-mode bits in MSTATUS represent HS-mode SSTATUS and S-mode bits in VSSTATUS represent VS-mode SSTATUS. When V=1, the S-mode bits in MSTATUS represent VS-mode SSTATUS and S-mode bits in VSSTATUS represetn HS-mode SSTATUS. When transitioning from V=0 to V=1 (or vice-versa), we swap the S-mode bits between MSTATUS and VSTATUS. This way the MSTATUS and SSTATUS emulation in Spike remain same for both HS-mode and VS-mode.

For 4), The MSTATUS and SSTATUS emulation ensures FS bits are writeable only when 'F' or 'D' extension is available (Refer, MSTATUS and SSTATUS emulation in set_csr() function). The SSTATUS can be viewed as restricted view of MSTATUS that's why set_csr() recursively calls itself for emulating SSTATUS. On other hand, VSSTATUS is a separate CSR so set_csr() emulates it separately and also checks for 'F' and 'D' extension availability.

I hope above addresses your queries. Let me know if I missed anything.

For questions or issues not related to this topic, please create another GitHub issue.

Regards,
Anup

tianhu-zeng, 2020-09-28:
@avpatel thanks for your reply. I have no doubt about mstatus.sd logic. Maybe I didn't describe my problem clearly.
1. SD bit is a RO field, but it's writable on vsstatus in  Spike. I think it's not make sense, right?
2. There are different conditions about writing FS field on mstatus/status/vstatus in Spike.
     1) mstatus: has_fs = supports_extentation('S') || supports_extentation('F') || supports_extentation('V').
     2) sstatus: can write directly.
     3) vsstatus: supports_extentation('F').
[image]
     Different conditions about writing FS field on mstatus/status/vstatus, does it make sense?
3. The following is my understanding about executing "csrw sstatus, x5" in VS mode.
     After executing this instruction, x5 is written to 'vsstatus' and shouldn't affect mstatus/sstatus. I write a case that execute "csrw sstatus, t1" in VS mode. I found that this writing affect 'mstatus/status' csrs, such as 'SD' bit. Following is the testcase and spike.log.
[image]

Regards,
Tianhu

avpatel, 2020-09-28:
1. Yes, SD bit of VSSTATUS should be treated as read-only in set_csr() implementation of processor.cc. This needs to be fixed.
2. The condition for writeability of MSTATUS.FS field in MSTATUS emulation is not updated as part of H-extension support in Spike (@aswaterman ??). The SSTATUS.FS and VSSTATUS.FS condition for writeabiltiy looks correct to me.
3. Your understanding is correct but we have taken CSR-swapping approach for MSTATUS/SSTATUS/VSSTATUS emulation. This approach provides functionally same behavior which you described. As-per this approach, when V=1 the SSTATUS represents VS-mode SSTATUS and HS-mode SSTATUS is saved in VSSTATUS CSR whereas when V=0 the SSTATUS represents HS-mode SSTATUS and VS-mode SSTATUS is saved in VSSTATUS CSR.

Regards,
Anup

aswaterman, 2020-09-28:
@avpatel I don't think the H-extension should affect the writability of mstatus.FS.  The condition for mstatus.FS' writability looks correct to me.

tianhu-zeng, 2020-09-29:
@avpatel @aswaterman thanks for your reply very much, it's very helpful to me. I'm very curious why [MSTATUS/SSTATUS/VSSTATUS].FS have different writability.

avpatel, 2020-10-01:
@aswaterman we have not modified mstatus.FS writability at time of adding H-extension to Spike. I guess we are good here. Can you check the writeablitiy condition for VSSTATUS.FS ?

aswaterman, 2020-10-09:
I see three problems:
- `set_virt` is treating the field like it's Boolean by doing things like `if ((state.mstatus & SSTATUS_FS) == SSTATUS_FS) state.vsstatus |= SSTATUS_FS;`.  This has the effect of turning the Initial and Clean states into Dirty.  Instead, the two-bit contents of the field should be copied, perhaps using the `get_field` macro.
- For the same reason, `set_virt` is setting SD=1 for FS=Initial or Clean, when in fact it should only do so for Dirty.
- It looks like the hypervisor is allowed to write the vsstatus.SD field.  Instead, on a vsstatus write, SD should be set to 1 iff FS/FS/XS is Dirty.  See, for comparison, how mstatus.SD is set: https://github.com/riscv/riscv-isa-sim/blob/f1c24eff543e6f41980993f41ae1ab5ab80a7340/riscv/processor.cc#L872

The handling of vsstatus.SD doesn't look correct: it seems that the hypervisor can simply write that bit, whereas it should always equal the expression (FS == Dirty || VS == Dirty).

tianhu-zeng, 2020-10-10:
@aswaterman @avpatel So, how should we deal with this issue?

avpatel, 2020-10-11:
I have submitted fix this https://github.com/riscv/riscv-isa-sim/pull/568

Regards,
Anup

tianhu-zeng, 2020-10-12:
@avpatel I have two questions. 
1. When are you going to push the fixing to the master? 
2. Why need check supports_extension("F") for vsstatus and didn't for sstatus?

aswaterman, 2020-10-12:
I think checking supports_extension("F") is actually not necessary because of the clause in the spec that FS exists if S-mode exists, even if the F extension doesn't.

tianhu-zeng, 2020-10-13:
@aswaterman Thanks for your reply. If so, Spike implementation match with Spec. Thanks again.

tianhu-zeng, 2020-10-26:
@avpatel When could you push the fixing version?

aswaterman, 2020-10-26:
Anup already made a PR. It is waiting on my review. Maybe in 2-3 days.

tianhu-zeng, 2020-10-26:
Got it. Does The Spike boot kvm or xvisor successfully?

avpatel, 2020-10-26:
Yes, both KVM and Xvisor work fine with the mentioned GitHub PR.

Regards,
Anup

tianhu-zeng, 2020-11-23:
Has this issue been fixed?

aswaterman, 2020-11-23:
Yes, it was fixed in #568.

closed 2020-11-23 by aswaterman