riscv-software-src/riscv-isa-sim #554: when mcounteren = 1 and hcounteren = 0, access user Counter/Timers csrs should report virtual instruction exception in VS/VU.

tianhu-zeng, 2020-09-28:
@avpatel when mcounteren = 1 and hcounteren = 0, access user Counter/Timers csrs should report virtual instruction exception in VS/VU mode. It seems that the logic is not complete in Spike.

aswaterman, 2020-09-28:
@avpatel

avpatel, 2020-09-28:
Spike only emulates INSTRET and CYCLE counters. The TIME counter and other counters are not emulated so for all such counters we will illegal instruction trap even in VS/VS mode.

For INSTRET and CYCLE counters, whenever HCOUNTER does not permit access to VS/VU mode will get virtual instruction trap. The get_csr() function in processor.cc is taking care of this case and jumping to throw_virtual label.

Regards,
Anup

tianhu-zeng, 2020-09-29:
@avpatel Thanks for your reply. So if we implement TIME counter and other counters, we need to modify Spike according to Spec by myself  to emulate those counters?

avpatel, 2020-10-01:
Yes, we sill have to modify Spike according spec whenever TIME and other counters are added.

Currently, TIME CSR is trap-n-emulated for S-mode by M-mode firmware.

aswaterman, 2020-10-01:
So are INSTRET and CYCLE handled correctly?  Can we close this issue?

tianhu-zeng, 2020-10-09:
Sorry for the late reply, last week we were on holiday. We will modify Spike according to Spec for all not emulated user timer/counter csrs on Spike. And we can close this issue.

closed 2020-10-09 by tianhu-zeng

aswaterman, 2020-10-09:
Spike should continue not implementing the time CSR and related CSRs, at least by default, since that matches many current implementations.

tianhu-zeng, 2020-10-09:
@aswaterman Got it. Thanks for your reply.