riscv-software-src/riscv-isa-sim #555: Spike crashes after around 1,5k of write_tohost: procedure calls.

dariuszst, 2020-09-28:
The problem occurs for the assembly code generated by riscv-dv generator. The code generated y riscv-dv is assembled with gcc toolchain compiler:
`riscv64-unknown-elf-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles riscv_jump_stress_test_1.S -I./../riscv-dv/user_extension -T./../riscv-dv/scripts/link.ld -o riscv_jump_stress_test_1.o -march=rv32i -mabi=ilp32`

and then simulate with a comannd:
` spike --log-commits --isa=rv32i -l riscv_jump_stress_test_1.o &> riscv_jump_stress_test.1.log`

There is an example of a program (and log file) for the rv32i target in the attached zip file ([riscv_jump_stress_test.zip](https://github.com/riscv/riscv-isa-sim/files/5292585/riscv_jump_stress_test.zip)).
It seems that write_tohost: procedure called in the ecall_handler: routine:

```
ecall_handler:    
                  la x25, _start
                  sw x0, 0(x25)
                  sw x1, 4(x25)
                 ...
                  sw x31, 124(x25)
                  la x3, write_tohost
                  jalr x0, x3, 0

...
write_tohost:     
                  sw gp, tohost, t5

_exit:            
                  j write_tohost
```

makes the spike stuck in a forever loop that ends with an internal error:
```
...
core   0: 0xffffffff8000737c (0x00001f17) auipc   t5, 0x1
3 0x8000737c (0x00001f17) x30 0x8000837c
core   0: 0xffffffff80007380 (0xc83f2223) sw      gp, -892(t5)
3 0x80007380 (0xc83f2223) mem 0x80008000 0x8000737c
core   0: 0xffffffff80007384 (0xff9ff06f) j       pc - 0x8
3 0x80007384 (0xff9ff06f)
core   0: 0xffffffff8000737c (0x00001f17) auipc   t5, 0x1
3 0x8000737c (0x00001f17) x30 0x8000837c
core   0: 0xffffffff80007380 (0xc83f2223) sw      gp, -892(t5)
3 0x80007380 (0xc83f2223) mem 0x80008000 0x8000737c
core   0: 0xffffffff80007384 (0xff9ff06f) j       pc - 0x8
3 0x80007384 (0xff9ff06f)
core   0: 0xffffffff8000737c (0x00001f17) auipc   t5, 0x1
3 0x8000737c (0x00001f17) x30 0x8000837c
core   0: 0xffffffff80007380 (0xc83f2223) sw      gp, -892(t5)
3 0x80007380 (0xc83f2223) mem 0x80008000 0x8000737c
core   0: 0xffffffff80007384 (0xff9ff06f) j       pc - 0x8
3 0x80007384 (0xff9ff06f)
core   0: 0xffffffff8000737c (0x00001f17) auipc   t5, 0x1
3 0x8000737c (0x00001f17) x30 0x8000837c
terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #14429289264837566231
```

aswaterman, 2020-09-28:
Note that `x3` is an alias for `gp`, so what you're doing is writing the address of `write_tohost` into `tohost`.  `tohost` can only be written with a certain set of values.  If you're trying to exit with success, you should write 1 to tohost.  If you're trying to exit with failure, write an odd-numbered value other than 1.

dariuszst, 2020-09-29:
OK, it seems that there is a bug in the SV riscv-dv generator that generates assembly code in which the GP register content is destroyed by overwriting it by using x3 register for jump purposes. 

However the output generated by spike still looks like unexpected application error and this behavior should be probably corrected.

aswaterman, 2020-09-29:
It is meant to be a fatal error, but the message could certainly be less terse.