riscv-software-src/riscv-isa-sim #557: Spike use 2 to represent PRV_HS. According to Spec, privilege=2 should be reserved.

tianhu-zeng, 2020-09-29:
@avpatel 
[image]
According to Spec defination, privilege=2 is reserved. But spike use PRV_HS(=2) to represent HS privilege.
In this case, if I write 2 to MSTATUS.MPP when hypervisor enable, it can be written to MSTATUS.MPP. It seems mismatch with Spec.

Regards,
Tianhu

aswaterman, 2020-09-29:
This is true: it should not be possible to set MPP=2.

tianhu-zeng, 2020-09-29:
@aswaterman thanks for your reply. Waiting for fixing.

avpatel, 2020-10-01:
@aswaterman the PRV_HS is a dummy mode and I am quite confident we are not setting PRV_HS to MSTATUS.MPP anywhere.

aswaterman, 2020-10-01:
@avpatel I understand it is used as a dummy mode.  The concern is that the RISC-V software could write the mstatus.MPP field to 2.  Doing so should result in MPP containing a value other than 2, but as written, it looks like MPP will become 2.

aswaterman, 2020-10-01:
I guess what I'm saying is I think we should revert the change to the `legalize_privilege` so that it doesn't recognize PRV_HS as a valid mode.  I don't think that will break any dummy uses of PRV_HS.

aswaterman, 2020-10-04:
Turns out the `legalize_privilege` change was introduced in #508, rather than part of your initial contribution.  I am going to revert #508 and close this issue.

commit 8664cd35 (2020-10-04): Revert "Fix legalize_privilege for extension H (#508)"

This reverts commit 8d860c190640e19e0f23a21d2479b4a36d13d342.

Fixes #557
commit 7dad2d45 (2020-10-04): It should never be possible to select MPP=2

This completes the fix in 8d860c190640e19e0f23a21d2479b4a36d13d342
to cover the hypervisor extension case as well.

Fixes #557
commit 8966775a (2020-10-04): It should never be possible to select MPP=2

This completes the fix in 8d860c190640e19e0f23a21d2479b4a36d13d342
to cover the hypervisor extension case as well.

Fixes #557
closed 2020-10-04 by aswaterman