riscv-software-src/riscv-isa-sim #570: Problem executing RSA with ISA RV32I

Marc43, 2020-10-14:
Hi,

I compiled spike with ISA RV32I (the default one), and I am executing the following instruction:

`sra x3, x3, x1` where `x3 = 0xcafecafe, x1 = 0x20`.

When executed, the value of `x3` is `0xffffffff`.

Why I am obtaining this result? Following the specs:

> SLL, SRL, and SRA perform logical left, logical right, and arithmetic right shifts on the value in register rs1 by the shift amount held in the lower 5 bits of register rs2.

If only the lower 5 bits of the value of register `rs2` are used, the value to be used should be `0x0` rather than `0x20`

I guess I am doing something wrong. But, what?

aswaterman, 2020-10-14:
I am guessing that Spike is actually running in RV64 mode... try `--isa=rv32i` on the command line when invoking it?

Marc43, 2020-10-14:
Yes, it was that. But, then, we should change the specified default field in the ISA:
```
Spike RISC-V ISA Simulator 1.0.1-dev

usage: spike [host options] <target program> [target options]
Host Options:
  -p<n>                 Simulate <n> processors [default 1]
  -m<n>                 Provide <n> MiB of target memory [default 2048]
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
                          at base addresses a and b (with 4 KiB alignment)
  -d                    Interactive debug mode
  -g                    Track histogram of PCs
  -l                    Generate a log of execution
  -h, --help            Print this help message
  -H                    Start halted, allowing a debugger to connect
  --isa=<name>          RISC-V ISA string [default RV32I]
  --priv=<m|mu|msu>     RISC-V privilege modes supported [default MSU]
  --varch=<name>        RISC-V Vector uArch string [default vlen:128,elen:64,slen:128]
  --pc=<address>        Override ELF entry point
  --hartids=<a,b,...>   Explicitly specify hartids, default is 0,1,...
  --ic=<S>:<W>:<B>      Instantiate a cache model with S sets,
  --dc=<S>:<W>:<B>        W ways, and B-byte blocks (with S and
  --l2=<S>:<W>:<B>        B both powers of 2).
  --device=<P,B,A>      Attach MMIO plugin device from an --extlib library
                          P -- Name of the MMIO plugin
                          B -- Base memory address of the device
                          A -- String arguments to pass to the plugin
                          This flag can be used multiple times.
                          The extlib flag for the library must come first.
  --log-cache-miss      Generate a log of cache miss
  --extension=<name>    Specify RoCC Extension
  --extlib=<name>       Shared library to load
                        This flag can be used multiple times.
  --rbb-port=<port>     Listen on <port> for remote bitbang connection
  --dump-dts            Print device tree string and exit
  --disable-dtb         Don't write the device tree blob into memory
  --kernel=<path>       Load kernel flat image into memory
  --initrd=<path>       Load kernel initrd into memory
  --bootargs=<args>     Provide custom bootargs for kernel [default: console=hvc0 earlycon=sbi]
  --real-time-clint     Increment clint time at real-time rate
  --dm-progsize=<words> Progsize for the debug module [default 2]
  --dm-sba=<bits>       Debug bus master supports up to <bits> wide accesses [default 0]
  --dm-auth             Debug module requires debugger to authenticate
  --dmi-rti=<n>         Number of Run-Test/Idle cycles required for a DMI access [default 0]
  --dm-abstract-rti=<n> Number of Run-Test/Idle cycles required for an abstract command to execute [default 0]
  --dm-no-hasel         Debug module supports hasel
  --dm-no-abstract-csr  Debug module won't support abstract to authenticate
  --dm-no-halt-groups   Debug module won't support halt groups
  --dm-no-impebreak     Debug module won't support implicit ebreak in program buffer
```
Thank you, for your time.

aswaterman, 2020-10-14:
Mysterious; the `--with-isa` configure option works fine over here.

Marc43, 2020-10-14:
Anyway I close the issue, thanks for your time!

closed 2020-10-14 by Marc43