riscv-software-src/riscv-isa-sim #571: MIP.SEIP should be writable if supervisor mode is implemented.

tianhu-zeng, 2020-10-15:
[image]
[image]

As pictures show. MIP.SEIP should be writable if supervisor mode is implemented, but spike didn't.

scottj97, 2020-11-11:
SEIP is part of [`supervisor_ints`](https://github.com/riscv/riscv-isa-sim/blob/956ef9ac3a66d2d6cda5df33a0e3f1c7d57ed0e0/riscv/processor.cc#L787).

scottj97, 2020-11-11:
Related: #139. Since Spike has no PLIC, mie.MEIE is not writable. 

But this case is a little different. Machine-mode software can simulate supervisor software interrupts, and it feels to me like that should work even without a PLIC in Spike.

scottj97, 2021-08-10:
This was fixed in #763

closed 2021-08-10 by aswaterman