riscv-software-src/riscv-isa-sim #574: enable/disable CLINT

udinator, 2020-10-16:
Hi,
For verification of the [Ibex](https://github.com/lowrisc/ibex) core, we have had to create a project-specific branch of Spike to add some local patches, specifically to remove part of the CLINT that raises an internal timer interrupt based on `mtime` and `mtimecmp` CSRs (as Ibex relies on an external PLIC instead of a CLINT).
This workaround was required since it appears that Spike always makes the assumption that there is a CLINT hooked onto the core with the timer registers. Since not all cores will have a CLINT (or will want to use it during simulation), we'd propose adding a command-line flag in the upstream codebase to enable/disable the CLINT in the ISS.
Would something like this be an acceptable solution to the issue?
Any feedback would be greatly appreciated!
Thanks.

aswaterman, 2020-10-17:
I think others have been adding this kind of fine-grained control using the DTS rather than the command-line, because the latter becomes quite unwieldly.  @khannaudit which would be your approach?

khannaudit, 2020-10-17:
I think it would be nice to make the DTS be the main method of specifying platform configuration knobs, and add only the most important knobs to the commandline (along with the knobs that the CLI already supports, for backward compatibility).

khannaudit, 2020-10-17:
In this case, the DTS.