riscv-software-src/riscv-isa-sim #578: Incorrect exception on guest address translation during VS table walk

jameshippisley, 2020-10-22:
According to H-extension specification:

https://github.com/riscv/riscv-isa-manual/blob/db3eeaf648dff66fab89b948a969833f951e841c/src/hypervisor.tex#L2429-L2433

"For a memory access made to support VS-stage address translation (such as to read/write a VS-level page table), permissions are checked as though for a load or store, not for the original access type. However, any exception is always reported for the original access type (instruction, load, or store/AMO)."

In spike:

s2xlate is called for the guest address translation stage of a memory access during a VS stage table walk here, with type LOAD

https://github.com/riscv/riscv-isa-sim/blob/b75aff9e5d132a16d9326bc0b3bbc724ae3c753c/riscv/mmu.cc#L390

s2xlate signals guest page faults with the supplied type:

https://github.com/riscv/riscv-isa-sim/blob/b75aff9e5d132a16d9326bc0b3bbc724ae3c753c/riscv/mmu.cc#L358-L363

It seems that to be consistent with the spec, and assuming the spec is considered correct, then the original access type for the access triggering the table walk also needs to be passed to s2xlate, and used in the switch statement above.

aswaterman, 2020-10-22:
Agreed - instead of unconditionally generating a load_guest_page_fault, the original access type should be passed to this switch statement.  cc @avpatel

avpatel, 2020-10-23:
Okay, I will submit a fix for this.

Regards,
Anup

aswaterman, 2020-10-24:
@jameshippisley can you confirm that #582 fixes the problem?

jameshippisley, 2020-10-25:
Yes, #582 fixes the problem.

closed 2020-10-26 by aswaterman