riscv-software-src/riscv-isa-sim #584: A confusion about sync vsstatus to mstatus.

tianhu-zeng, 2020-10-27:
1. In Spike, when executing mret/sret, it will sync vsstatus.{sd,fs,vs} to mstatus.{sd,fs,vs}. 
2. The Spike checks mstatus.fs when access fp csr.

Confusions:
1. Why bonds sync operations to mret/sret? 
2. In Spec, accessing fp csr need checking sstatus.fs and vsstatus.fs in VS-mode. But in Spike, due to the sync operation, it just check mstatus.fs. It seems that the Spike mismatch with the Spec. 
[image]
[image]
[image]

tianhu-zeng, 2020-10-27:
@avpatel @aswaterman

aswaterman, 2020-10-27:
You don't need to @-mention me, since I don't maintain this code.

tianhu-zeng, 2020-10-29:
@avpatel could you have a look about this issue? thanks.

avpatel, 2020-10-29:
The generating the illegal-instruction trap considering both SSTATUS.FS and VSSTATUS.FS was missed out whereas updating dirty state is taken care by set_virt() which is called to change processor virt state.

Linux, KVM and Xvisor, don't see any issues because all these software set SSTATUS.FS and VSSTATUS.FS to initial state (not disabled).

tianhu-zeng, 2020-10-29:
understand, thanks for your reply.

closed 2020-10-29 by tianhu-zeng

francis4096, 2021-02-06:
> The generating the illegal-instruction trap considering both SSTATUS.FS and VSSTATUS.FS was missed out whereas updating dirty state is taken care by set_virt() which is called to change processor [...]

@avpatel, Do you mean that checking VSSTATUS.FS needs to be added to "require_fp" macro? 
I tried to modify the require_fp marco to like this.
#define require_fp require((((STATE.mstatus & MSTATUS_FS) != 0) & (STATE.v == 0)) | (((STATE.mstatus & MSTATUS_FS) != 0) & ((STATE.vsstatus & SSTATUS_FS) != 0) & STATE.v))

But kvm complaints illegal instruction after Run /virt/init as init process. The sepc address is "fld ft0,0(a0)".
------------------------------------
[    2.601769] Freeing unused kernel memory: 2136K
[    5.024622] Run /virt/init as init process
[    8.980426] kvm [49]: VCPU exit error -14
[    8.980700] kvm [49]: SEPC=0xffffffe000005282 SSTATUS=0x8000000200006120 HSTATUS=0x2002001c0
[    8.981270] kvm [49]: SCAUSE=0x2 STVAL=0x53007 HTVAL=0x0 HTINST=0x0
KVM_RUN failed: Bad address
------------------------------------
Do you have any idea about this issue?

francis4096, 2021-02-16:
@avpatel ，Could you help to check this issue?

avpatel, 2021-02-16:
Apologies for the delay.

Your require_fp() macro looks incorrect because it uses "&" for logical-AND and "|" for logical-OR

Can try the below definition of require_fp() macro ?

```
#define require_fp require((((STATE.mstatus & MSTATUS_FS) != 0) && (STATE.v == 0)) || (((STATE.mstatus & MSTATUS_FS) != 0) && ((STATE.vsstatus & SSTATUS_FS) != 0) && STATE.v))
```

aswaterman, 2021-02-17:
@avpatel / @francis4096  Let me know when the PR is fixed and I should merge it.

francis4096, 2021-02-17:
> Apologies for the delay.
>
> Your require_fp() macro looks incorrect because it uses "&" for logical-AND and "|" for logical-OR
>
> Can try the below definition of require_fp() macro ?
>
> ```
> #define [...]

@avpatel  Thank you for correction. I fixed it in PR. But the KVM issue still exists. I think there may be something wrong with KVM software. Could we merge this PR to spike first?