riscv-software-src/riscv-isa-sim #585: Any plan to add windows support for spike

fanghuaqi, 2020-10-27:
Hi there,

     Spike is actually a great functional model for riscv processor starter, but I found that there is no windows support for it, any plan to add support for windows?
      If we have windows support for it, it will be great for user who works in windows to use spike to emulate risc-v core. And it will be easier for tool developer to integrate spike in IDE tools.

Thanks
Huaqi

aswaterman, 2020-10-27:
Windows support is not in the plan of record.

We strive for POSIX compliance, but even Cygwin doesn't provide some of the necessary APIs.  This is especially problematic given that part of Spike is a Linux syscall emulation layer.

I haven't tried WSL.

fanghuaqi, 2020-10-27:
Thank you for the information @aswaterman