riscv-software-src/riscv-isa-sim #587: Incompatible with rvv-intrinsics gcc.

edre, 2020-10-28:
As with #531, I'm getting some version incompatibility when trying to run binaries produced with the [rvv-intrinsic branch of gcc](https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-intrinsic). It's not documented clearly what version of the vector spec that gcc branch or spike is currently implementing. I tried a recent spike version (1d30acb) and the version for 0.9 suggested in #531 (67b7edd), but in both cases got an illegal instruction error at the first vsetvl.

edre, 2020-10-28:
And to clarify what encoding is causing the problems, the instruction bits from the dump are `01057557`.

aswaterman, 2020-10-29:
Two things to check: that you're providing the V extension (e.g. `--isa=rv64gcv`) and that your boot code is enabling the V extension (by setting mstatus.VS to nonzero).

edre, 2020-10-29:
Ah, thanks for bringing the mstatus.VS field to my attention. This looks like a new change since 0.9, but the latest riscv-v-spec doesn't seem to cover its semantics in much detail. The spec references states Initial, Clean, Dirty, and you imply that zero is an Off state. Which bit value is Initial? What happens to the vector state if it gets set to Off? Are vector subroutines expected or encouraged to aggressively turn off the vector status when they're done with it (much earlier versions of the v spec did this)?

edre, 2020-10-29:
Reading this a little more carefully, mstatus is obviously in the M layer, and by boot code, you meant kernel boot. Since I'm running spike through pk, I assume this means I need to update my pk version.

aswaterman, 2020-10-29:
Yeah, IIRC a newer pk will suffice.

The VS field has the same encoding as the FS field, which you can read about in the priv spec.  Setting it to anything nonzero will suffice for these purposes.

edre, 2020-10-29:
Updating pk resolved the issue. Can't wait for 1.0 ratification so I can stop filing dumb issues about these things :)

closed 2020-10-29 by edre

aswaterman, 2020-10-29:
Cool, glad that worked.