riscv-software-src/riscv-isa-sim #589: Not able to simulate hello program

rahulraveendran15-coder, 2020-11-03:
Hi, 
I am getting the below error when running the command 
`spike pk hello_world`

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)
```
How to overcome this?
I want to run with 32imc

My Tool chain paths are as below:
```

export PATH=${PATH}:/home/rr/.local/opt/riscv32/lowrisc-toolchain-gcc-rv32imc-20200904-1/bin
export RISCV_TOOLCHAIN=$HOME/.local/opt/riscv32/lowrisc-toolchain-gcc-rv32imc-20200904-1/
export RISCV_GCC="$RISCV_TOOLCHAIN/bin/riscv32-unknown-elf-gcc"
```
The build commands I tried is below

```
./configure --prefix=$RISCV_TOOLCHAIN/riscv32-unknown-elf --enable-commitlog --enable-misaligned 
./configure --prefix=$RISCV_TOOLCHAIN--enable-commitlog --enable-misaligned (tried this command also later)
make
sudo make install
```
```
riscv32-unknown-elf-gcc -o hello hello.c
spike pk hello
```
Can anyone pleae let me know where I am missing something?

emelcher, 2021-04-12:
Hi,
my environment for `riscv32-unknown-elf-gcc` is
`export PATH=$PATH:$RISCV/bin `
for pk installation I used:
`../configure --prefix=$RISCV --with-arch=rv32i --host=riscv32-unknown-elf --target=riscv32-unknown-elf`
for spike installation:
`../configure --prefix=$RISCV --with-isa=rv32i --target=riscv32-unknown-elf --enable-histogram `
After installation implemented this workaround:
```
mkdir -p $RISCV/riscv64-unknown-elf/bin
cp $RISCV/riscv32-unknown-elf/bin/pk $RISCV/riscv64-unknown-elf/bin/
```
Now it's working:
```
$ riscv32-unknown-elf-gcc -o hello hello.c
$ spike pk hello
bbl loader
Hello world!
```

emelcher, 2021-04-15:
for spike installation use:

../configure --prefix=$RISCV --with-isa=rv32i --target=riscv32-unknown-elf --enable-histogram CXXFLAGS="-DTARGET_ARCH=\\\\\\\"riscv32-unknown-elf\\\\\\\""

The workaround from my previous comment is not needed any more.