riscv-software-src/riscv-isa-sim #590: Does the Spike simulator support N-Extension?

SaiVK, 2020-11-07:
Hello Everyone
   I am trying to make use of U-Mode Interrupts in an application that I am developing. In the _README.md_  file, under _Extensions Supported_,  **_N-Extension_** was not mentioned. Is there any development branch of Spike where I could find the N-Extension supported?

Please guide me on this issue.

Thanking you
Sai

aswaterman, 2020-11-08:
No, and the current plan is to deprecate the N extension in favor of "bare S-mode"--i.e. for embedded systems that need to take interrupts into multiple privilege modes, the less-privileged code will run in S-mode.

SaiVK, 2020-11-09:
Hello @aswaterman 
Oh ok. Interesting to hear that. Is there any development draft relating to the bare S-mode? And any idea when it will be released?

Thank you so much
Sai

aswaterman, 2020-11-09:
It pretty much already exists. Just run your privileged code in M-mode and
your unprivileged code in S-mode.

SaiVK, 2020-11-09:
Hello @aswaterman 
Oh ok. Deprecating the N version means that, User mode interrupts/traps are not going be supported? Did I understand it correctly. If that is the case, then how would User mode interrupts/traps be supported?

Thank you so much
Sai

sorear, 2020-11-09:
Can you elaborate on what you're requesting?  "user mode" is not an absolute concept, if you have a PMP system you can run unprivileged code in S-mode.

SaiVK, 2020-11-09:
By User mode, I mean the U-mode. Assuming a system has all three modes: M, S, U. If I remember correctly, delegating interrupts to U-mode come under the N-extension. So, when N-Extension is going to get deprecated, does that mean, U-Mode interrupts won't be supported? I got a little confused.

Please correct me if I am wrong.

Thank you so much
Sai

sorear, 2020-11-09:
That's correct.  If you have M, S, and U as implemented modes, interrupts can be taken in M or S mode.

Note that S-mode memory access is restricted by the configured PMP registers, and if you set the TVM bit of mstatus to 1 S-mode will also be restricted by any configured page tables.

Note also that virtual memory support is optional even if S-mode is implemented.  If satp is hardwired to 0 (bare), the amount of logic required to implement S-mode for interrupts is essentially the same as that required by the N draft.

SaiVK, 2020-11-10:
Ok, I understand your point. So when all 3 modes are present, interrupts would be supported only in M & S mode only, not in U mode.
But, I had come across a paper: [Donky](https://www.usenix.org/conference/usenixsecurity20/presentation/schrammel), whose implementation of Intra-Process Isolation based on Page Table Domains is based on N-Extension(U-Mode interrupts). So, does that mean the idea proposed in this paper would need to be re-implemented once N-Extension is deprecated?
Also, Given that all three modes are present in a system, it might be good to support U-Mode interrupts, since they don't have to switch privileges on an interrupt/trap. 

Thank you so much
Sai

sorear, 2020-11-10:
The N extension does not solve security problems, it creates them (riscv/riscv-isa-manual#557); it's frustrating to see *security researchers* lured in by this attractive nuisance.  So yes.

A secure mechanism for user-level traps in a 3-level system running a normal supervisor is possible to design, but it would necessarily look rather different from today's N.

SaiVK, 2020-11-10:
I see. So is there any future plan to support user-level traps in a 3-level system running a normal supervisor?

Thank you so much 
Sai

sorear, 2020-11-10:
I'm not personally aware of such plans, but I'm not in the best position to know.

SaiVK, 2020-11-10:
Oh ok. Thank you so much @sorear for patiently answering my questions.

-Sai