riscv-software-src/riscv-isa-sim #600: Spike boot from address 0x1000 and then jump to reset vector

panda1628, 2020-11-18:
This is the trace of a RISC-V compliance test. The reset vector address is at 0xffffffff80000000
but spike starts at 0x1000 and execute a few instructions before jump to the reset vector.
I believe it match rocket chip which has a boot ROM at 0x1000

I cannot find these instructions in the *.elf file, so I assume they are build -in spike, is it true?
Also, is there any way I can skip them and start from reset vector?

I run the same *.elf with OVPsim and it starts from reset vector directly.

core 0: 0x0000000000001000 (0x00000297) auipc t0, 0x0
core 0: 3 0x00001000 (0x00000297) x 5 0x00001000
core 0: 0x0000000000001004 (0x02028593) addi a1, t0, 32
core 0: 3 0x00001004 (0x02028593) x11 0x00001020
core 0: 0x0000000000001008 (0xf1402573) csrr a0, mhartid
core 0: 3 0x00001008 (0xf1402573) x10 0x00000000
core 0: 0x000000000000100c (0x0182a283) lw t0, 24(t0)
core 0: 3 0x0000100c (0x0182a283) x 5 0x80000000 mem 0x00001018
core 0: 0x0000000000001010 (0x00028067) jr t0
core 0: 3 0x00001010 (0x00028067)
core 0: 0xffffffff80000000 (0x04c0006f) j pc + 0x4c
core 0: 3 0x80000000 (0x04c0006f)
core 0: 0xffffffff8000004c (0xf1402573) csrr a0, mhartid
core 0: 3 0x8000004c (0xf1402573) x10 0x00000000
core 0: 0xffffffff80000050 (0x00051063) bnez a0, pc + 0
core 0: 3 0x80000050 (0x00051063)

aswaterman, 2020-11-19:
They are built into spike; see here: https://github.com/riscv/riscv-isa-sim/blob/fce242a5d495db731eee6571916399d10ec531e9/riscv/sim.cc#L290

panda1628, 2020-11-19:
Thanks

May be we shall add an option to bypass it and starts from the reset code?

aswaterman, 2020-11-19:
Many real processors will also execute some machine-level code in ROM before jumping to instruction RAM. So I don’t view Spike’s behavior as sufficiently problematic.

Currently you can sort of do what you’re suggesting with the —disable-dtb option, but in that case you must put a RAM at address 0x1000 (see the —help for the -m option) and use that as the entry point.