riscv-software-src/riscv-isa-sim #604: A question about GVA update condition.

tianhu-zeng, 2020-11-27:
In Spec 5.2.1 "Field GVA (Guest Virtual Address) is written by the implementation whenever a trap is taken into HS-mode. For any trap (access fault, page fault, or guest-page fault) that writes a guest virtual address to stval, GVA is set to 1. For any other trap into HS-mode, GVA is set to 0."
In Spec 5.4.1 "Field GVA (Guest Virtual Address) is written by the implementation whenever a trap is taken into M-mode. For any trap (access fault, page fault, or guest-page fault) that writes a guest virtual address to mtval, GVA is set to 1. For any other trap into M-mode, GVA is set to 0."

The function take_trap in the Spike:
[image]

My understanding is that only trap (access fault, page fault, or guest-page fault)  from v=1, gva can be set to 1. But in the spike, all trap(v=1/v=0) from mmu will set gva, even for address misaligned. It seems that the spike implementation differ from the Spec. Could you have a look about this issue, thanks.

@avpatel

avpatel, 2020-11-27:
This is a good catch. This one slipped through the cracks because most hypervisors don't depend on HSTATUS.GVA bit.

I will work on a fix for this.

Regards,
Anup

avpatel, 2020-11-27:
I have submitted PR https://github.com/riscv/riscv-isa-sim/pull/605 to fix this issue.

Regards,
Anup

aswaterman, 2020-11-30:
#605 has been merged.

closed 2020-11-30 by aswaterman

461659642, 2021-06-10:
Hi, avpatel and waterman,
From the Spec 5.4.1 "Field GVA (Guest Virtual Address) is written by the implementation whenever a trap is taken into M-mode. For any trap (access fault, page fault, or guest-page fault) **that writes a guest virtual address to mtval**, GVA is set to 1. For any other trap into M-mode, GVA is set to 0."
My understanding is that, the GVA field should also be set to 1 when these faults of hypervisor load/store occur, because these instructions trigger access/page/guestpage fault, then the mtval is also updated to guest virtual address, is right?

461659642, 2021-06-10:
@avpatel

aswaterman, 2021-06-10:
@461659642 Please change your github account to indicate your name and affiliation before using this issue tracker anymore.

DizhaoWei, 2021-06-10:
@aswaterman I'm sorry, it was an oversight on my part.