riscv-software-src/riscv-isa-sim #612: Can HS-mode take vssip/vseip/vstip interrupt?

tianhu-zeng, 2020-12-04:
My understanding is that vs interrupt trap to the specific mode which based  on mideleg and hideleg about vseip/vstip/vssip.
All vs interrupts were traped to VS mode on Spike. Are there special considerations for this implementation? Thanks!
[image]

@avpatel

avpatel, 2020-12-18:
The bits corresponding to VS interrupts are hardwired to 1 in MIDELEG CSR so M-mode cannot never take VS interrupt. This is in accordance with section "5.4.2 Machine Interrupt Delegation Register (mideleg)".

The H-extension spec certainly allows HS-mode to take VS interrupts because VS interrupt bits are writeable in HIDELEG CSR. I think "~MIP_VS_MASK" in take_interrupt() should be replaced with "~state.hideleg". Let send a fix for this.

avpatel, 2020-12-18:
I have add fix to the following PR (https://github.com/riscv/riscv-isa-sim/pull/618)

closed 2021-12-07 by scottj97