riscv-software-src/riscv-isa-sim #617: Spike misaligned lr/sc behavior does not match spec

jameshippisley, 2020-12-16:
According to the following section of the privileged spec, when a regular load or store for an address can complete without signaling an exception, lr/sc/atomics must not signal a misaligned address exception for that address. Instead, they should signal an access fault exception. Assuming I have understood the spec correctly, this is to inform the kernel that it should not attempt to emulate the misaligned lr/sc/atomic, since it cannot depend on regular accesses to the same address to occur under the same mutex.

https://github.com/riscv/riscv-isa-manual/blob/6cd9eec691b6bf8a5762a8806bb0a07e28b84818/src/machine.tex#L2810-L2837

Spike however does signal a misaligned load/store exception in this case.

https://github.com/riscv/riscv-isa-sim/blob/4b04d715298c7c0542badf30828f65d1f02e1c1d/riscv/mmu.h#L94-L101

https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/lr_w.h#L2

As a side note, I have observed that riscvOVPsim also signals a misaligned load/store exception in this case.

I want to check whether the spike behavior is considered a bug, or whether I should read the spec as saying that this behavior is optional. The non-privileged spec states this as a choice:

https://github.com/riscv/riscv-isa-manual/blob/db3eeaf648dff66fab89b948a969833f951e841c/src/a.tex#L151-L158

At least some of the wording in the privileged spec section I linked to implies that signaling a misaligned address exception in this case is not allowed.

aswaterman, 2020-12-16:
I think Spike is doing the right thing when misaligned-access support is disabled (as is the default), since regular loads and stores will also raise address-misaligned exceptions.  Agreed?

When misaligned-access support is enabled, I agree that an access fault should instead be raised here, for the reason you mentioned.  Would you like to make a PR, or shall I fix it?

jameshippisley, 2020-12-17:
Agreed that the default misaligned-access disabled behavior is correct.

I can make a PR, although it may be a few days until I get it done.

aswaterman, 2020-12-17:
OK, I'll try to take care of it tomorrow.

aswaterman, 2020-12-19:
@jameshippisley please confirm that 438999600c459663f23c2156371743181fe5ef77 appears to fix the problem.

closed 2020-12-19 by aswaterman

jameshippisley, 2020-12-19:
Confirming that this fixes the problem.

commit 43899960 (2020-12-19): If misaligned accesses are enabled, throw access fault on misaligned LR/SC

See #617 for discussion.

Resolves #617.