riscv-software-src/riscv-isa-sim #621: Multi cycle instruction

qs132, 2020-12-22:
hello,
     If i want to customize a multi cycle instruction，what should i do? Do the same thing as custom single cycle instructions?Can spike simulate the number of running cycles？I tried to use rdcycle instruction to get the number of running cycles. But I don't know how to deal with multi cycle instructions. By the way,Is the number of cycles obtained with spike accurate？
Thank you very much!
qin

aswaterman, 2020-12-22:
Spike is only a functional simulator and is not suitable for microarchitecture experiments.  You might want to take a look at gem5, though I don't personally have any experience with those tools.

If you are really just trying to count cycles, you could strategically insert some printfs then post-process the resulting output.

qs132, 2020-12-22:
ok
Thank you!

oaltinay, 2021-04-22:
Hi, I implemented an algorithm to Spike so that it counts the clock cycle of instructions that do not take 1 clock.  [In my repo](https://github.com/oaltinay/riscv-isa-sim/blob/master/riscv/clock_cycles_map.h), you can add instructions that do not take 1 clock cycle and Spike will update the CSR cycle according to the clock cycle specified.