riscv-software-src/riscv-isa-sim #622: Big-endian supports breaks `fesvr` library support

zarubaf, 2020-12-28:
When compiling against `libfesvr` the big-endian support (#575) breaks the `fesvr` library functionality. In particular, the header file `memif.h` depends on `byteorder.h`:

https://github.com/riscv/riscv-isa-sim/blob/032a68c3e94b0d315207a5a42251f836185b5d01/fesvr/memif.h#L8

which is not installed by `make install`. Adding `byteorder.h` to the list of installed header files fails because it relies on `autotool` generated `config.h` which should not be distributed/installed.

aswaterman, 2020-12-28:
I don't think we need a separate library because this is only a header file.  Shouldn't it just suffice to move byteorder.h into fesvr/ and add it to the install-hdrs list?  Can you confirm that 0f264a174ed6ec1cf18bf7d34aed79cdf2deb66d fixes this?

closed 2020-12-28 by aswaterman

zarubaf, 2020-12-29:
`byteorder.h` depends on `config.h` (generated during build time by `configure`), so https://github.com/riscv/riscv-isa-sim/commit/0f264a174ed6ec1cf18bf7d34aed79cdf2deb66d, unfortunately, doesn't help.

So any compilation against `fesvr` will cause the following error:
```
In file included from /Users/fz/riscv/bin/../include/fesvr/htif.h:6:
In file included from /Users/fz/riscv/bin/../include/fesvr/memif.h:8:
/Users/fz/riscv/bin/../include/fesvr/byteorder.h:6:10: fatal error: 'config.h' file not found
#include "config.h"
         ^~~~~~~~~~
1 error generated.
```

aswaterman, 2020-12-29:
Indeed.  Installing config.h is a clean solution.  This time I actually tested that I don't get a compile error when including fesvr/htif.h :)

34de90b78eddd49c880ae46cfbcf274822e9c5da

commit 0f264a17 (2020-12-28): Install fesvr/byteorder.h to fix #622