riscv-software-src/riscv-isa-sim #624: Run multi harts on spike and get running info

suzhaow, 2020-12-29:
It is possible to run multicore/mulithart on spike using -p(num).
I can get the complete simulation information of core1, but I cannot get the complete simulation information of core0 (the second half of the information is missing) in the output log file.
And the simulation information of the two cores are stored in the same log file.
Here is my execution command:
<path_var>/spike -p2 --log-commits --isa=<variant> -m0x40000000:0x50000000 -l <elf>
Is there something i am missing ?
Thanks.

chihminchao, 2020-12-29:
No,  the above command is correct. The log of each hart should be printed out.
What does the "the second half of the information is missing"  mean ? Could you provide an snip ?

suzhaow, 2020-12-30:
Is there any way to print core0 and core1 information separately?
And this is the info of the spike simulation. 
The information of core0 and core1 are printed interspersedly. 
Sorry, I made a mistake(I can get the complete simulation information of core0, not core1).
Finally, core0 jumps into the loop, and core1 only prints to a certain section of the program(core1 also jumped into the loop in the RLT simulation).
[riscv_custom_instr_test.0.log](https://github.com/riscv/riscv-isa-sim/files/5752771/riscv_custom_instr_test.0.log)
Thanks.

chihminchao, 2020-12-30:
It is a little weird. 
`core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0`  ->this is shown by -l
`3 0x0000000000001000 (0x00000297) x 5 0x0000000000001000`  -> this is shown by commitlog but it lack the core number information.
What is the spike version ?

suzhaow, 2020-12-30:
Sorry, I don't know much about it.
And I just git the latest riscv-isa-sim file from GitHub to overwrite my original riscv-isa-sim file yesterday.
Do I need to modify something in the file?

suzhaow, 2020-12-30:
Oh I see. And I had follow the rvv: commitlog: get hartid directly about riscv/execute.cc. But it doesn't work.
[https://github.com/riscv/riscv-isa-sim/commit/f37ebaca770d91a90b83fe2be1ae61028ae69152?branch=f37ebaca770d91a90b83fe2be1ae61028ae69152&diff=split](url)

suzhaow, 2021-01-04:
Hello, I encountered a new problem while updating the new version.
1. Using spike --log_commits in the new version will cause an error: unrecognized option --log_commits

2. RISCV instruction execution error occurs in the new version (spike --priv=msu),
core0: 0x000000004000024c (0x140c9cf3) csrrw s9,sscratch s9
core0: exception trap illegal_instruction,epc 0x000000004000024c
core0: tval 0x00000000140c9cf3
The same problem occurs when reading sepc and scause information.

Is there something i am missing ?
Thanks.

closed 2021-01-04 by suzhaow