riscv-software-src/riscv-isa-sim #633: BUG: Unable to experiment with spike plugins

maroueneboubakri, 2021-01-20:
Hi,

I tried to play with spike plugins but encountered the following error at first use which seems to be an issue on parsing --device argument.

```
$ spike --extlib ./uart/uart.so --device serial,0x02100000,0 -l --isa rv32imac ../riscv-probe/build/bin/rv32imac/spike/hello
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Aborted (core dumped)
```

Thanks
Maro

maroueneboubakri, 2021-01-20:
Based on the provided example, and for some reason, the name **should** be _test_mmio_plugin_

stevobailey, 2022-01-11:
Hi Maro, were you able to get this to work? I'm able to compile and run the example from [here](https://github.com/vexingcodes/spike-plugin), but it's not actually acting like an MMIO region. I see stores printed, but I can't get data to actually be stored somewhere and loaded correctly when loads are called. Do you have a working example of an MMIO plugin that just acts like a memory region?