riscv-software-src/riscv-isa-sim #644: Adding support for mac instruction in spike:

subhajit26, 2021-02-12:
Hi,
I am trying add support for custom instruction called mac in spike. This is how I have tried to add the functionality in a file
called mac.h. I have made all necessary changes. I have already added mod instruction before and that seems to work pretty well. But I believe in this case the issue is with the assignment of the registers. 
```
require_extension('M');
WRITE_RD(sext_xlen(RD + (RS1 * RS2));
```
But while building spike again I facing errors. Please see the errors below. Can you please suggest where am I going wrong and how to fix it?
```
In file included from mac.cc:9:0:
../riscv/insns/mac.h:2:0: error: unterminated argument list invoking macro "WRITE_RD"
 WRITE_RD(sext_xlen(RD + (RS1 * RS2));
 ^
In file included from mac.cc:18:0:
../riscv/insns/mac.h:2:0: error: unterminated argument list invoking macro "WRITE_RD"
 WRITE_RD(sext_xlen(RD + (RS1 * RS2));
 ^
In file included from mac.cc:9:0:
../riscv/insns/mac.h: In function ‘reg_t rv32_mac(processor_t*, insn_t, reg_t)’:
../riscv/insns/mac.h:2:1: error: ‘WRITE_RD’ was not declared in this scope
 WRITE_RD(sext_xlen(RD + (RS1 * RS2));
 ^
mac.cc:10:3: error: expected ‘;’ before ‘trace_opcode’
   trace_opcode(p,  MATCH_MAC, insn);
   ^
In file included from mac.cc:18:0:
../riscv/insns/mac.h: In function ‘reg_t rv64_mac(processor_t*, insn_t, reg_t)’:
../riscv/insns/mac.h:2:1: error: ‘WRITE_RD’ was not declared in this scope
 WRITE_RD(sext_xlen(RD + (RS1 * RS2));
 ^
mac.cc:19:3: error: expected ‘;’ before ‘trace_opcode’
   trace_opcode(p,  MATCH_MAC, insn);
   ^
make: *** [mac.o] Error 1
```

aswaterman, 2021-02-12:
You're missing a closing parenthesis on the second line.

subhajit26, 2021-02-15:
Hi Andrew,
                    Thank you for pointing the mistake. I have done the correction and was successfully able to implement the mac instruction.

closed 2021-02-22 by subhajit26