riscv-software-src/riscv-isa-sim #650: PMP behavior when PMP is off and in U-mode

mmxsrup, 2021-02-19:
## Observed Behavior
I found that spike keeps working without throwing an exception in u-mode when transitioning from m mode to u mode without setting PMP. I noticed this because [ibex](https://github.com/lowRISC/ibex) and spike behave differently. I have already submitted this to the [Ibex issue](https://github.com/lowRISC/ibex/issues/1277). Is it the expected behavior in U-mode not to generate a pmp error when pmp is not set?

There are the following comments in the Ibex issue.
https://github.com/lowRISC/ibex/issues/1277#issuecomment-781970209

## Steps to reproduce the issue
```c
#include "simple_system_common.h"

#define MSTATUS_MPP_M (3 << 11)
#define MSTATUS_MPP_U (0 << 11)

void test() {
  return;
}

int main(int argc, char **argv) {
  // from M mode to U mode
  uint32_t mstatus;
  asm volatile("csrr %0, mstatus" : "=r"(mstatus));
  mstatus = (mstatus & ~MSTATUS_MPP_M) | MSTATUS_MPP_U;
  asm volatile(
      "csrw mstatus, %0;"
      "csrw mepc, %1;"
      "mret;"
      :
      : "r"(mstatus), "r"(test));
  return 0;
}
```
Compile the test program and  run this on spike.
```
spike --isa=rv32imc --log-commits -d -m0x10000:0x30000,0x100000:0x100000 --priv=mu code.elf
```

From the spike log, you can see that after the mret instruction, it continues to operate without raising an exception.
```
core   0: >>>>  main
core   0: 0x00000000001003da (0x300027f3) csrr    a5, mstatus
core   0: 3 0x001003da (0x300027f3) x15 0x00000000
: 
core   0: 0x00000000001003de (0x00007779) c.lui   a4, 0xffffe
core   0: 3 0x001003de (0x7779) x14 0xffffe000
: 
core   0: 0x00000000001003e0 (0x7ff70713) addi    a4, a4, 2047
core   0: 3 0x001003e0 (0x7ff70713) x14 0xffffe7ff
: 
core   0: 0x00000000001003e4 (0x00008ff9) c.and   a5, a4
core   0: 3 0x001003e4 (0x8ff9) x15 0x00000000
: 
core   0: 0x00000000001003e6 (0x00000717) auipc   a4, 0x0
core   0: 3 0x001003e6 (0x00000717) x14 0x001003e6
: 
core   0: 0x00000000001003ea (0xf6670713) addi    a4, a4, -154
core   0: 3 0x001003ea (0xf6670713) x14 0x0010034c
: 
core   0: 0x00000000001003ee (0x30079073) csrw    mstatus, a5
core   0: 3 0x001003ee (0x30079073) c768_mstatus 0x00000000
: 
core   0: 0x00000000001003f2 (0x34171073) csrw    mepc, a4
core   0: 3 0x001003f2 (0x34171073) c833_mepc 0x0010034c
: 
core   0: 0x00000000001003f6 (0x30200073) mret
core   0: 3 0x001003f6 (0x30200073) c768_mstatus 0x00000080
: 
core   0: >>>>  test
core   0: 0x000000000010034c (0x00008082) ret
core   0: 0 0x0010034c (0x8082)
: 
core   0: 0x00000000001003c8 (0x000202b7) lui     t0, 0x20
core   0: 0 0x001003c8 (0x000202b7) x 5 0x00020000
: 
core   0: 0x00000000001003cc (0x000002a1) c.addi  t0, 8
core   0: 0 0x001003cc (0x02a1) x 5 0x00020008
```

aswaterman, 2021-02-19:
Spike was written before PMP existed, and so to preserve compatibility with existing software, Spike resets the highest-numbered PMP register to a value that permits U-mode accesses.

Since the reset values for the PMP registers are platform-defined, both Spike and Ibex are doing something technically valid here.  Portable software should initialize the PMP registers, rather than assuming they have a particular value.

mmxsrup, 2021-02-22:
Understood. Thank you for your polite comment.

closed 2021-02-22 by mmxsrup