riscv-software-src/riscv-isa-sim #651: SIMD instructions support:

subhajit26, 2021-02-22:
Hi,
     I want to add SIMD instruction support to the spike so I was wondering if I can use the below issue as a reference:
`Support RISC-V p-ext-proposal v0.9 SIMD Data Processing Instructions #572`

As of now the riscv-isa-sim repo that I have cloned does not have p-ext support but I can see in this issue some updation to the files needed to add support to the spike has been done. So , do I I have to clone and build the spike from the below link:

"I have created a [p-ext](url) branch. I will continue to update p-ext instructions on that branch."

Please suggest as I am trying add support for the add8 instruction.

subhajit26, 2021-02-22:
Hi,
I am trying to add the support for add8 in spike. I can see that it is already updated.
This is the code that I am trying to run:

```
#include <stdio.h>
#include <stddef.h>
#include <stdint.h>

int main() {
    int32_t a[4]={1,2,3,4};
    int32_t b[4]={1,2,3,4};
    int32_t c[4];
    int i;
    
    for (i=0 ; i < 4; i++){
    asm volatile
    (
    "add8   %[z], %[x], %[y]\n\t"
    : [z] "=r" (c[i])
    : [x] "r" (a[i]), [y] "r" (b[i])
    ); 
        //c[i] = a[i] + b[i];
        
        printf("result_vector:%d\n",c[i]);
    }
    return 0;
}
```
I have already added the support for add8 in riscv-gnu -toolchain with updated match and mask bit. The code is properly compiled with riscv32-unknown-elf-gcc.
#define MATCH_ADD8 0x48000077
#define MASK_ADD8 0xfe00707f

But when I run the executable in spike I get the below error:

```
bl loader
z  00000000 ra 000100bc sp 7fbeed20 gp 00020a90
tp 00000000 t0 00010286 t1 0000000f t2 00000000
s0 7fbeed70 s1 00000000 a0 00000001 a1 7fbeed74
a2 00000001 a3 7fbeed60 a4 00000001 a5 00000001
a6 00000000 a7 00000000 s2 00000000 s3 00000000
s4 00000000 s5 00000000 s6 00000000 s7 00000000
s8 00000000 s9 00000000 sA 00000000 sB 00000000
t3 00000000 t4 00000000 t5 00000000 t6 00000000
pc 00010178 va 48f70777 insn       48f70777 sr 80046020
An illegal instruction was executed!
```
I an running spike after building it in p-ext branch.
I have a doubt regarding the Define type in disasm.cc. I do not see the instrcutiono name add8 in the updated disasm.cc. Please suggest how to fix the issue.

when I run the same code with add instruction in asm volatile it works but I want the code to work with add8 simd instruction.

hope51607, 2021-02-22:
Hi, Do you use --isa to enable p extension like --isa=RV32IMAFDCP?

By the way, I have completed all rvp v0.9.2 instructions support and created a new PR #637

subhajit26, 2021-02-22:
Hi,
      I am using this command:
spike --isa=RV32IMC  $(which pk) <path and filename> .
Do I need to use P in --isa to enable P extension.
--isa to enable p extension like --isa=RV32IMAFDCP? 
I am bit new to this packed extension so I am not much aware of the enabling.
But if I add it in isa , I believe that I also need to build the ricv-elf toolchain with P extension enabled. As of now I have RV32IMC built toolchain.

subhajit26, 2021-02-22:
Hi,
         In the link you have given I can see the you have updated the disasm.cc file. But I do not see the Define of add8 instruction which comes under DISASM_8_AND_16_RINSN(add). Is it not needed like this defining structure: DEFINE_RTYPE(cras16);
```
DISASM_8_AND_16_RINSN(add);
  DISASM_8_AND_16_RINSN(radd);
  DISASM_8_AND_16_RINSN(uradd);
  DISASM_8_AND_16_RINSN(kadd);
  DISASM_8_AND_16_RINSN(ukadd);
  DISASM_8_AND_16_RINSN(sub);
  DISASM_8_AND_16_RINSN(rsub);
  DISASM_8_AND_16_RINSN(ursub);
  DISASM_8_AND_16_RINSN(ksub);
  DISASM_8_AND_16_RINSN(uksub);
  DEFINE_RTYPE(cras16);
  DEFINE_RTYPE(rcras16);
  DEFINE_RTYPE(urcras16);
  DEFINE_RTYPE(kcras16);
  DEFINE_RTYPE(ukcras16);
  DEFINE_RTYPE(crsa16);
  DEFINE_RTYPE(rcrsa16);
  DEFINE_RTYPE(urcrsa16);
  DEFINE_RTYPE(kcrsa16);
  DEFINE_RTYPE(ukcrsa16);
  DEFINE_RTYPE(stas16);
  DEFINE_RTYPE(rstas16);
  DEFINE_RTYPE(urstas16);
  DEFINE_RTYPE(kstas16);
  DEFINE_RTYPE(ukstas16);
  DEFINE_RTYPE(stsa16);
  DEFINE_RTYPE(rstsa16);
  DEFINE_RTYPE(urstsa16);
  DEFINE_RTYPE(kstsa16);
  DEFINE_RTYPE(ukstsa16);
```

hope51607, 2021-02-22:
Hi, 
Spike option--isa just tells spike which extension instructions can be simulated.
Therefore, if you want to simulate add8, you only need to add "P" to your spike --isa option
like --isa=RV32IMCP

subhajit26, 2021-02-22:
Hi Chung,
                    Thanks for the information. I will try it and let you know if I face any issue.

hope51607, 2021-02-22:
> Hi,
> In the link you have given I can see the you have updated the disasm.cc file. But I do not see the Define of add8 instruction which comes under DISASM_8_AND_16_RINSN(add). Is it not needed like [...]

The define of `DISASM_8_AND_16_RINSN` is also in `disasm.cc `
```C++
#define DISASM_8_AND_16_RINSN(code) \
  DEFINE_RTYPE(code##8); \
  DEFINE_RTYPE(code##16);
```
So `DISASM_8_AND_16_RINSN(add)` define disasm of add8 and add16 at the same time.

subhajit26, 2021-02-22:
Hi Chung,
                   I got the point "So DISASM_8_AND_16_RINSN(add) define disasm of add8 and add16 at the same time."
And the code also ran with the P extension enabled. Thanks a lot again for the clarification.

subhajit26, 2021-02-23:
Hi ChunPing,
                 Do we have any support for loading multiple data from memory to a big register at once with single load instruction.
Let say loading 4 (8-bit data) to 32-bit register. Is there any work around in the packed extension to allow this sort of SIMD operation.

hope51607, 2021-02-24:
As far as I know, in p-spec v0.9.2, only have packing instruction which pack two registers data to another register.

I suggest that this kind of question can be asked in https://github.com/riscv/riscv-p-spec or https://lists.riscv.org/g/tech-p-ext/topics
and you can get more professional reply.

subhajit26, 2021-02-24:
Hi Chun,
                I just need to clear few doubts regarding the add16  since I am going to use it as a custom instruction for the FFT algo.
```
P_LOOP(16, {
  pd = ps1 + ps2;
})
```
what do you mean by the P_LOOP. How is the functionality implemented.

1. Basically,  I will use this instruction for parallel addition of the two 16 bit values of a complex number and pack  it into 32 bit register.
2. Along with the parallel addition I also want to get one step right shift like the functionality below:
   `pd = (ps1 + ps2)>>1;`
Is it possible to implement this functionality within one instruction. Separate shift instruction exists but If I want to get this in one instruction which all files do I need to change. Please suggest.

hope51607, 2021-02-25:
> what do you mean by the P_LOOP. How is the functionality implemented.

P_LOOP is implemented in decode.h 

> Is it possible to implement this functionality within one instruction. Separate shift instruction exists but If I want to get this in one instruction which all files do I need to change. Please suggest.

Maybe what you want is radd16.