riscv-software-src/riscv-isa-sim #652: RV32EMC support

subhajit26, 2021-02-22:
Hi,
    Right now the spike has rv32imc support. I am being able to run the executable file properly from tool-chain build with 
    RV32IMC support.

    I have also build tool-chain with RV32EMC  and now I am trying to run the executable in spike but it gives me error so I am not 
    sure whether spike has RV32EMC support or not.

I am running the below commands:
This is for the pk stuff:
`export PATH=$PATH:/home/subhajit/riscv-gnu-toolchain/build_emc/riscv32-unknown-elf/bin`
Executing the file:
`spike --isa=RV32EMC $(which pk) /home/subhajit/riscv-gnu-toolchain/build_emc/bin/mac`
It shows me error:
`error: bad --isa option 'RV32EMC'. unsupported extension 'e'`

For IMC I have used this command to configure and rest all the other commands worked properly:
`../configure --enable-commitlog --enable-misaligned --prefix=$RISCV`

Please suggest. Do I need to build the spike separately again for the emc support (if so, please give me the configure command.) or is there anything I need to change in the execution command mentioned above.

WangQing911, 2021-05-08:
hello， I met the same problem. Have you solved your problem ？

subhajit26, 2021-05-08:
Hi,
     Unfortunately, I was not able to get setup the support for RV32EMC. Some where in the previous issues they have claimed that EMC works with spike but I was not successful in adding that support which you can see in the above thread. I am fully sure but I think spike still does not have EMC support.

aswaterman, 2021-05-08:
Spike doesn't directly support RV32E.  However, RV32E software should generally work correctly on an RV32I implementation (assuming it doesn't attempt to cause an exception by accessing x16-x31).

jmonesti, 2022-01-06:
Hello Andrew,
Currently facing the same issue.
I agree that "RV32E software should generally work correctly on an RV32I implementation"
Unfortunately, this also has side-effect on register MISA:
Refering to Volume II: RISC-V Privileged Architectures V20211203 p.17
4 | E | RV32E base ISA
8 | I | RV32I/64I/128I base ISA
We observe differences between RTL & the reference model in our verification testbench.
I think this is worth fixing, would you accept a pull request in processor_t::par_isa_string() ?

Regards,

scottj97, 2022-01-06:
If Spike's going to show E in misa then it should actually implement E fully, including the exceptions on accessing x16-x31.

aswaterman, 2022-01-06:
What Scott said.  In the meantime, I suggest you make a private fork of Spike and hack it for your cosimulation purposes.

aswaterman, 2022-01-07:
I've thrown together an RV32E prototype: https://github.com/riscv-software-src/riscv-isa-sim/pull/899

aswaterman, 2022-01-10:
Resolved via #899

closed 2022-01-10 by aswaterman