riscv-software-src/riscv-isa-sim #662: Error: illegal Instruction executed.

subhajit26, 2021-03-01:
Hi,
       I am trying to implement a custom instruction "absd" which will give me the absolute value of the difference between 2 signed 32 bit values. Here is the functionality defined in spike below: Please let me know if there is any problem in the functionality:
```
if(sext_xlen(RS1) >= sext_xlen(RS2))
  WRITE_RD(sext_xlen(sext_xlen(RS1) - sext_xlen(RS2)));
else
  WRITE_RD(sext_xlen(sext_xlen(RS2) - sext_xlen(RS1)));
```
I have added the instruction in encoding.h, disasm.cc and riscv.mk.in and also added the support for the same in riscv-elf tool-chain under instruction class I with 3 register allocation:
I want to put the custom instruction to replace the implementation of the specific part of the code.
```
int32_t abs_diff;
            if (x >= y) {
                abs_diff = x - y;// custom instruction
            } else {
                abs_diff = y - x;// custom instruction
            }
```
Custom implementation:
```
asm volatile
    (
    "absd   %[z], %[x], %[y]\n\t"
    : [z] "=r" (abs_diff)
    : [x] "r" (x), [y] "r" (y)
    );

```

oaltinay, 2021-04-22:
Could you please share the error you got? Did "absd" instruction generated correctly in the assembly code?

subhajit26, 2021-04-22:
Hey,
        I got the instruction working in a very simple way instead of the way I was trying in above thread.

closed 2021-12-07 by scottj97