riscv-software-src/riscv-isa-sim #672: WFI continues without any interrupt

IanJiangICT, 2021-03-10:
Description: 
1. Core 0 halts at a WFI
1. Core 1 takes turn
1. Core 0 continues without any interrupt

Note: Parameter INTERLEAVE of Spike is changed from 5000 to 16 to make the running shorter.

Run log
```
$ spike -l -p2 test-wfi 2> run-spike.log
$ less run-spike.log
warning: tohost and fromhost symbols not in ELF; can't communicate with target
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   0: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
core   0: 0x0000000000001010 (0x00028067) jr      t0
core   0: >>>>  
core   0: 0x0000000080000000 (0x300022f3) csrr    t0, mstatus
core   0: 0x0000000080000004 (0x0082e293) ori     t0, t0, 8
core   0: 0x0000000080000008 (0x30029073) csrw    mstatus, t0
core   0: >>>>  switch_thread
core   0: 0x000000008000000c (0xf14022f3) csrr    t0, mhartid
core   0: 0x0000000080000010 (0x02028763) beqz    t0, pc + 46
core   0: >>>>  thread_master
core   0: 0x000000008000003e (0x00028293) mv      t0, t0
core   0: 0x0000000080000042 (0x10500073) wfi
core   0: 3 0x0000000080000042 (0x10500073)
core   1: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   1: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   1: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   1: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
core   1: 0x0000000000001010 (0x00028067) jr      t0
core   1: >>>>  
core   1: 0x0000000080000000 (0x300022f3) csrr    t0, mstatus
core   1: 0x0000000080000004 (0x0082e293) ori     t0, t0, 8
core   1: 0x0000000080000008 (0x30029073) csrw    mstatus, t0
core   1: >>>>  switch_thread
core   1: 0x000000008000000c (0xf14022f3) csrr    t0, mhartid
core   1: 0x0000000080000010 (0x02028763) beqz    t0, pc + 46
core   1: 0x0000000080000014 (0x0000a009) c.j     pc + 2
core   1: >>>>  thread_slave
core   1: 0x0000000080000016 (0x00050513) mv      a0, a0
core   1: 0x000000008000001a (0x00000585) c.addi  a1, 1
core   1: 0x000000008000001c (0x00000609) c.addi  a2, 2
core   1: 0x000000008000001e (0x0000068d) c.addi  a3, 3
core   1: 0x0000000080000020 (0x00000711) c.addi  a4, 4
core   0: 0x0000000080000046 (0x00000305) c.addi  t1, 1
core   0: 0x0000000080000048 (0x00000389) c.addi  t2, 2
core   0: 0x000000008000004a (0x00000e0d) c.addi  t3, 3
core   0: 0x000000008000004c (0x00000001) c.nop
core   0: 0x000000008000004e (0x0000a009) c.j     pc + 2
core   0: >>>>  spike_exit
core   0: 0x0000000080000050 (0x00000001) c.nop
core   0: 0x0000000080000052 (0x00000001) c.nop
core   0: 0x0000000080000054 (0x0000bff5) c.j     pc - 4
core   0: >>>>  spike_exit
core   0: 0x0000000080000050 (0x00000001) c.nop
core   0: 0x0000000080000052 (0x00000001) c.nop
core   0: 0x0000000080000054 (0x0000bff5) c.j     pc - 4
```

Test source code:
```
$ cat test-wfi.S 
.section .text
.align 2
.global _start

_start:
	csrr	t0, mstatus
	ori	t0, t0, 0x8
	csrw	mstatus, t0

switch_thread:
	csrr	t0, mhartid
	beqz	t0, thread_master
	j	thread_slave

thread_slave:
	addi	a0, a0, 0
	addi	a1, a1, 1
	addi	a2, a2, 2
	addi	a3, a3, 3
	addi	a4, a4, 4
	addi	a5, a5, 5
	addi	a6, a6, 6
	addi	a7, a7, 7
	addi	a0, a0, 0
	addi	a1, a1, 1
	addi	a2, a2, 2
	addi	a3, a3, 3
	addi	a4, a4, 4
	addi	a5, a5, 5
	addi	a6, a6, 6
	addi	a7, a7, 7
	nop
	j spike_exit

thread_master:
	addi	t0, t0, 0
	wfi
	addi	t1, t1, 1
	addi	t2, t2, 2
	addi	t3, t3, 3
	nop
	j spike_exit

spike_exit:
	nop
	nop
	j	spike_exit	
```
Tuning Spike
```
$ git diff
diff --git a/riscv/sim.h b/riscv/sim.h
index 1c47ce7..b5a850e 100644
--- a/riscv/sim.h
+++ b/riscv/sim.h
@@ -77,7 +77,7 @@ private:
 
   processor_t* get_core(const std::string& i);
   void step(size_t n); // step through simulation
-  static const size_t INTERLEAVE = 5000;
+  static const size_t INTERLEAVE = 16;
   static const size_t INSNS_PER_RTC_TICK = 100; // 10 MHz clock for 1 BIPS core
   static const size_t CPU_HZ = 1000000000; // 1GHz CPU
   size_t current_step;
```

IanJiangICT, 2021-03-10:
I checked the source code of Spike, but found nothing about checking interrupt for WFI.
```
riscv/decode.h
 303 #define wfi() \
 304   do { set_pc_and_serialize(npc); \
 305        npc = PC_SERIALIZE_WFI; \
 306        throw wait_for_interrupt_t(); \
 307      } while(0)

riscv/execute.cc
369     catch (wait_for_interrupt_t &t)
370     { 
371       // Return to the outer simulation loop, which gives other devices/harts a
372       // chance to generate interrupts.
373       //
374       // In the debug ROM this prevents us from wasting time looping, but also
375       // allows us to switch to other threads only once per idle loop in case
376       // there is activity.
377       n = ++instret;
378     }

```

scottj97, 2021-03-25:
It is legal for WFI to resume without waiting for an interrupt. This is an implementation choice, not a bug.

aswaterman, 2021-03-25:
Relevant clause in the ISA spec is here: https://github.com/riscv/riscv-isa-manual/blob/bf971f91f95e9c9cece7f9a137b3e276a66332f5/src/machine.tex#L2499

closed 2021-03-25 by aswaterman