riscv-software-src/riscv-isa-sim #673: Spike segfaults when running the attached elf

sthiruva, 2021-03-10:
I am getting a segfault while running spike since last 2 days.

```
   |177       reg_t npc;                                                                                                                                                                                          |
   |178                                                                                                                                                                                                           |
   |179       try {                                                                                                                                                                                               |
  >|180         npc = fetch.func(p, fetch.insn, pc);                                                                                                                                                              |
   |181         if (npc != PC_SERIALIZE_BEFORE) {                                                                                                                                                                 |
   |182                                                                                                                                                                                                           |
   |183     #ifdef RISCV_ENABLE_COMMITLOG                                                                                                                                                                         |
   |184           if (p->get_log_commits_enabled()) {                                                                                                                                                             |
   |185             commit_log_print_insn(p, pc, fetch.insn);                                                                                                                                                     |
   |186           }  

(gdb) bt
#0  0xffffffffffffffff in ?? ()
#1  0x000000000046b6b9 in execute_insn (p=p@entry=0x7ffff7f48010, pc=pc@entry=2237735196, fetch=...) at ../riscv/execute.cc:180
#2  0x000000000046dcba in processor_t::step (this=0x7ffff7f48010, n=4359, n@entry=5000) at ./icache.h:143
#3  0x000000000047ff3e in sim_t::step (this=this@entry=0x7fffffffd190, n=n@entry=5000) at ../riscv/sim.cc:194
#4  0x000000000044b833 in sim_t::main (this=0x7fffffffd190) at ../riscv/sim.cc:175
#5  0x00007ffff6d9c190 in ?? () from /lib64/libc.so.6
#6  0x0000000000b1cf70 in ?? ()
#7  0x0000000000000000 in ?? ()
(gdb) up
#1  0x000000000046b6b9 in execute_insn (p=p@entry=0x7ffff7f48010, pc=pc@entry=2237735196, fetch=...) at ../riscv/execute.cc:180
(gdb)
```
The program counter points to the following instruction. It is an illegal form of JALR instruction.

```
000000008561211c <start_SNP_Trap_Excp_Illegal_RV32_JALR_BadFunct3_run_block0_2883_19_1>:
    8561211c:   0f0333e7                0xf0333e7
```
The failure can be reproduced by running the command given below with the file in the attachment.

$STING_ROOT/tools/spike/bin/spike -p2 -m0x80000000:0x20000000 --isa=RV64IMAFDC sting.elf.spike
[sting.elf.spike.gz](https://github.com/riscv/riscv-isa-sim/files/6117843/sting.elf.spike.gz)

aswaterman, 2021-03-10:
@ben-marshall Looks like #649 is causing this.  I will look into it.

aswaterman, 2021-03-11:
Thanks for the bug report, @sthiruva.  It'll be fixed shortly.

aswaterman, 2021-03-11:
@ben-marshall The issue is that sometimes we were creating `insn_desc_t` structures without initializing the `insn_desc_t::archen` field.  It's a quick fix, but I took a different route: rather than having the new `archen` field at all, we can use the null-ness of the `insn_desc_t::rv32` and `insn_desc_t::rv64` fields to connote the same information.

commit a31ac28a (2021-03-11): Fix and refactor RV32-only and RV64-only instruction handling

We can use the nullness of pointers to indicate RV32-only or RV64-only
instructions, rather than a new field dedicated to that purpsoe.

The bug fix is that the new field wasn't always initialized; now,
it doesn't need to be.

Resolves #673