riscv-software-src/riscv-isa-sim #675: Vector 0.9 inst print

suzhaow, 2021-03-14:
I used the spike version 67b7edd and it worked.
But the behavior in spike is strange.
Here is the code:

`3 0x00000000400011ac (0x23dd) x 7 0xffffffffffcd8775
core   0: 0x00000000400011ae (0x94840457) vsll.vv v8, v8, v8, v0.t
3 0x00000000400011ae (0x94840457) e64 m2 l4
core   0: 0x00000000400011b2 (0x3b803457) vslideup.vi v8, v24, 0
3 0x00000000400011b2 (0x3b803457) e64 m2 l4 v 8 0x00000000000000000000000000000000 v 9 0x00000000000000000000000000000000                                                           
core   0: 0x00000000400011b6 (0x41c3dc3b) sraw    s8, t2, t3
3 0x00000000400011b6 (0x41c3dc3b) x24 0xffffffffffffffff
core   0: 0x00000000400011ba (0x2c014457) vxor.vx v8, v0, sp, v0.t
3 0x00000000400011ba (0x2c014457) e64 m2 l4`

Some write-back data are not printed, such as vsll, vxor. 
It may be because the value written back is the same as the original register value. 
Is there any way to make it print out ? 
Thanks!

chihminchao, 2021-03-14:
Commitlog only record write operation and print the written value no matter what value is written.  The missing destination register may result from the masking by v0.t

suzhaow, 2021-03-15:
Yeah I got it.
By the way, are there any rules for the order in which the vector command prints the rd register?
Some instructions are printed out of order occasionally. Here is the code :
`
Out of order
core   0: 0x0000000040001e54 (0x9f01b457) vmv4r.v v8, v16                                 
3 0x0000000040001e54 (0x9f01b457) e64 m2 l4 v11 0x000000000575657a000000000e98af02 v10 0x00000000d203d12d000000000f598197 v 8 0x00000000000000007ff800000000000e v 9 0xd7fa95459f0a701e0000000000000000

By order
core   0: 0x0000000040001e7a (0x9e01b857) vmv4r.v v16, v0
3 0x0000000040001e7a (0x9e01b857) e64 m2 l4 v19 0x000000000575657a000000000e98af02 v18 0x00000000d203d12d000000000f598197 v17 0x7fffffff7fc000007fffffff7fc00000 v16 0x7fffffff7fc000007fffffff7fc00000

core   0: 0x00000000400025aa (0x9f81b857) vmv4r.v v16, v24                                
3 0x00000000400025aa (0x9f81b857) e64 m2 l4 v19 0x000000000575657a000000000e98af02 v18 0x00000000d203d12d000000000f598197 v17 0x00000000000000207ff8000000000000 v16 0x7ff80000000000007ff800000000000f

`
Thanks

chihminchao, 2021-03-17:
The order is not guaranteed since it is stored in unordered_map. The order is decided by the C++ STL implementation.

closed 2021-04-14 by suzhaow