riscv-software-src/riscv-isa-sim #677: Can spike run dynamically linked multi-thread application

qqjinger, 2021-03-16:
Hi, 
    Recently, I compile coremark application with multi-thread options in beremetal mode. When running on spike, an error occurs. Error message is that "*** FAILED *** (tohost = 1337)". I have several questio as below:
    1) Does spike support multi-thread application?
    2) Does spike suppoet dynamically linked application?
    3) Does spike support running  baremetal application?
    I really appreciate your help. Thanks!

aswaterman, 2021-03-16:
Spike is capable of running valid RISC-V programs in general, but to support dynamic linking and multithreading, you need an OS (or a bare-metal software stack with many of the capabilities of an OS).  Running Linux on Spike is the most direct way.

qqjinger, 2021-03-17:
Thank you for your reply.