riscv-software-src/riscv-isa-sim #682: Sign extension of xperm insns

marcfedorow, 2021-03-25:
https://github.com/riscv/riscv-isa-sim/blob/634b7eb46e4d1b38f280ffeec9f0ebd8b579c17f/riscv/insns/xperm_b.h#L2

A problem occurs when simulating 32 bit registers:
```asm
li t1 -1;           # 0xffffffffffffffff is stored in t1
li t2 0;            # 0x0000000000000000 is stored in t1
xperm.b t0, t1, t2; # 0x00000000ffffffff is stored in t0
li t1 0xffffffff;   # 0xffffffffffffffff is stored in t1
bne t0, t1, fail;   # jumps to fail
```
Same problem could be observed on some p-extension insns.
This happens because 32-bit registers are stored in 64-bit variables.

I think this problem can be fixed in one of the following ways:
- Every insn's output should be sign-extended (my current way)
- Every insn's output should be zero-extended
- High-order bits should be ignored by ```beq```, ```bne```, etc.
The last option could be implemented easily but such a decision will probably have a bad effect on the performance.

Please clarify which path is preferable -- or what is my mistake.

aswaterman, 2021-03-25:
This appears to be a bug in the implementation of xperm.n, xperm.b, and xperm.h; they should call `sext_xlen` to sign-extend when in RV32 mode.  I'll push a fix.

cc @cliffordwolf

aswaterman, 2021-03-25:
Fixed in 4403faee253339c88d077abf6ef7539febad3e30

closed 2021-03-25 by aswaterman