riscv-software-src/riscv-isa-sim #686: Make error

muhammadsalmanafzal, 2021-03-30:
While trying to build spike, I got the following error:
```
./spike_dasm/spike_dasm_option_parser.cc:1:1: error: expected unqualified-id before ‘.’ token
 ../fesvr/option_parser.cc
 ^
make: *** [spike_dasm_option_parser.o] Error 1

```
Is there an issue with the options that I gave while configuration or there is an error with one of the files?

zhongchengyong, 2021-03-30:
@muhammadsalmanafzal The ./spike_dasm/spike_dasm_option_parser.cc is a symbolic file, I guess you get the source code by download zip, try git clone to get source code and build.

muhammadsalmanafzal, 2021-03-31:
> @muhammadsalmanafzal The ./spike_dasm/spike_dasm_option_parser.cc is a symbolic file, I guess you get the source code by download zip, try git clone to get source code and build.

No, I didn't downloaded it. Anyways, I tried it again but still failed at the same step.

muhammadsalmanafzal, 2021-04-02:
@zhongchengyong can you kindly have a look at it? I'll appreciate it. thanks.

zhongchengyong, 2021-04-08:
@muhammadsalmanafzal Sorry for the late reply, did you get source code by 'git clone' command? What's your operating system? You should use a linux distribution to compile it.

muhammadsalmanafzal, 2021-04-10:
@zhongchengyong It worked. Anyways, yes, I was using the ```git clone``` command from the very start. And I'm using CentOS 7 with kernel version ```5.11.8-1.el7.elrepo.x86_64``` on VMware player. 

What I did different now is that I cloned into a local folder of VM (before I was cloning into the shared folder that I mount on guest-CentOS7 with host-Windows 10). Maybe, the error was coming due to some shared folder disturbance. Although, my RISCV toolchain is still located in the shared folder and during ```../configure```, I put the ```--prefix``` to the mounted folder.

If you feel like this issue can be closed now, kindly tell me or if you would like to look further into matter, I can produce the results for you and you can take a look at the log. Thanks for helping.

closed 2021-04-14 by muhammadsalmanafzal