riscv-software-src/riscv-isa-sim #691: Running Opentitan ELF files on spike

Rahul-Kande, 2021-04-06:
Hello, 

I am trying to simulate the spike with elf files generated from the Opentitan SoC software  (https://github.com/lowRISC/opentitan/tree/master/sw/device/examples/hello_world). 
I have tried to run it in both baremetal and pk methods. Unfortunately, neither of them work.

Here are how I run them:
1. baremetal method:
      $spike --isa=rv32imc -l hello_world_sim_verilator.elf
      $Error: terminate called after throwing an instance of 'trap_load_access_fault'
2. pk method:
      $spike --isa=rv32imc -l pk hello_world_sim_verilator.elf
        or
      $spike --isa=rv32imc -l pk -p hello_world_sim_verilator.elf
The pk method keeps running and never terminates. When I check its log file, it causes several exceptions such as "trap_load_page_fault, trap_store_page_fault, etc."

Please help me solve this issue.
Thank you

-----------------------------------------------------------------------------------------------
Note: Here are the details about my setup:

- I use the lowRISC-provided prebuilt toolchain https://docs.opentitan.org/doc/ug/install_instructions/#software-development
riscv32-unknown-elf-gcc (crosstool-NG 1.24.0.41-475d72f) 9.2.0
- System information: 
        I am running on a VM Linux User-VirtualBox 5.8.0-45-generic #51~20.04.1-Ubuntu SMP Tue Feb 23 13:46:31 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
- Spike:
        build configuration: ../configure --prefix=$prebuilt_RISCV
- pk: https://github.com/lowRISC/riscv-pk
        build configuration: ../configure --prefix=$prebuilt_RISCV/riscv32-unknow-elf --host=riscv32-unknow-elf --with-abi=ilp32 --with-arch=rv32imc

silabs-hfegran, 2021-05-31:
Probably a Spike memory map conflict with your binary (Spike has hard coded ROMs and other simulator specific memory ranges that probably interferes), try something like the following for bare metal. (Spike expects binaries to conform to a certain memory map, the PK should normally take care of this for hosted applications, but if you want to run generic binaries on bare metal a few changes are needed)

What you need to do to get things running bare metal is probably to modify the memory map in spike to relocate the hard coded regions or modify your linker script for the opentitan binary. (Spike has debug reserved memory from 0x0-0xFFF, Boot rom at 0x1000-0x1FFF, and DRAM by default at 0x8000_0000 (IIRC)- There are some more defined regions in the spike source, but I don't remember these on the top of my head. The boot rom is disabled altogether by using the --disable-dtb flag)

Furthermore you'll probably need to define the memory ranges in your binary on the spike command line as well, example below should get you started. 

spike --isa=rv32imc --disable-dtb -l -m<start_addr_mem1>:<len_mem1>,<start_addr_mem2>:<len_mem2> hello_world_sim_verilator.elf

Manahil14, 2022-06-02:
How to decide these ranges of memory addresses? It causes access exceptions

spike --isa=rv32imc --disable-dtb -l -m hello
spike: argument required for option -m
Try 'spike --help' for more information.
akeana@akeana-Latitude-3520:~/proj/tools/spike-demo$ spike --isa=rv32imc --disable-dtb -l -m2048 hello
Access exception occurred while loading payload hello:
Memory address 0x135b0 is invalid
akeana@akeana-Latitude-3520:~/proj/tools/spike-demo$ spike --isa=rv32imc --disable-dtb -l -m0x00000000 hello
spike: ../riscv/cfg.h:44: mem_cfg_t::mem_cfg_t(reg_t, reg_t): Assertion `(size % PGSIZE == 0) && (base % PGSIZE == 0) && (base + size > base)' failed.
Aborted (core dumped)
akeana@akeana-Latitude-3520:~/proj/tools/spike-demo$ spike --isa=rv32imc --disable-dtb -l -m0x00000000:0xFFF hello
Warning: the memory at  [0x0, 0xFFE] has been realigned
to the 4 KiB page size: [0x0, 0xFFF]
Access exception occurred while loading payload hello:
Memory address 0x135b0 is invalid
akeana@akeana-Latitude-3520:~/proj/tools/spike-demo$ spike --isa=rv32imc --disable-dtb -l -m0x00000000:0xFFE hello
Warning: the memory at  [0x0, 0xFFD] has been realigned
to the 4 KiB page size: [0x0, 0xFFF]
Access exception occurred while loading payload hello:
Memory address 0x135b0 is invalid

scottj97, 2022-06-02:
We don't know how your ELF file was compiled, so we have no way to know where all the target program is expecting memories. Based on the error messages above, it apparently expects memory at 0x135b0, so you could try something like `-m0x10000:0x10000` to put memory at addresses 0x10000-0x1ffff.

Likely you will need several more besides that one.