riscv-software-src/riscv-isa-sim #695: rvp: strange ranges

marcfedorow, 2021-04-13:
Among other things #694 fixes uclipNN insns.
It seems strange that clipping range is stored in int64_t while it needs only intNN_t for sclipNN and uintNN_t for uclipNN.
It could (and I think should) be calculated as
```cpp
uint##NN##_t uint_max = (1 << immBu) - 1;
```
for uclipNN and
```cpp
int##NN##_t int_max = (1 << immBu) - 1;
int##NN##_t int_min = -int_max - 1;
```
, where `1 << B = NN` (so there is no UB with `1 << 32` because imm5u stores numbers from 0 to 31).

another oddity I noticed is excess memory usage for cycle ranges, e.g.
```
sreg_t len = 32 / BIT; \
for (sreg_t i = len - 1; i >= 0; --i) { \
```
It seems useless for me because 32 is a 32-bit integer and it's result of it's division by another 32-bit integer is obviously a 32-bit integer. While 32 is not a very big number, result could be stored in a `uint_fast8_t` or `auto`.

I thing both `decoder.h` and `insns/*` could be checked on excess range integers to ensure there is no other out-of-range bugs.

hope51607, 2021-07-08:
I agree with the point you mentioned about clipNN and cycle ranges.
Maybe you can send a PR to fix them?

scottj97, 2021-12-07:
Is this now fixed? By #740?

marcfedorow, 2021-12-07:
> Is this now fixed? By #740?

I believe there are no more wrongly calculated values.
Despite this, there is still no strict logic in used types.

scottj97, 2021-12-07:
Does that mean there's still an issue? Or can we close this?

marcfedorow, 2021-12-09:
I will close it but probably I will refactor it some day.
Of course this is not a priority task since I am not that one who responsible for P-ext.

closed 2021-12-09 by marcfedorow