riscv-software-src/riscv-isa-sim #702: instret == cycle ?

ALLap97, 2021-04-25:
Hi!
I use the following code to get the cycles:
```
unsigned long read_cycles(void)
{
  unsigned long cycles;
  asm volatile ("rdcycle %0" : "=r" (cycles));
  return cycles;
}
```
And i use the following code to get the instret count:
```long long read_instrets(void)
{
  long long instrets;
  asm volatile ("rdinstret %0" : "=r" (instrets));
  return instrets;
}
```
but i get the same result，the cycles == intructions count?
```
spike --isa=rv64gcv pk blur
bbl loader
read the image!
[800 x 600]
Scalar Instrets = 16708149
RVV Instrets = 5121300
RVVm8 instrets = 647065

spike --isa=rv64gcv pk blur2
bbl loader
read the image!
[800 x 600]
Scalar cycles = 16708149
RVV cycles = 5121300
RVVm8 cycles = 647065
```
Is the number of cycles obtained by spike inaccurate？

aswaterman, 2021-04-26:
Spike doesn’t model the microarchitecture, so cycle counts are meaningless. Instead, we arbitrarily say that Spike models harts with IPC=1, so cycle and instret are equal.

ALLap97, 2021-04-26:
yeah，I check the internal implementation of cycle CSR in Spike，thanks a lot!

closed 2021-04-26 by ALLap97