riscv-software-src/riscv-isa-sim #703: PMP granularity and pmpaddr CSR implementation

aprnath, 2021-04-26:
Hi,

If we have a granularity greater than 4bytes - say 64 bytes - then we notice that Spike is not implementing the unused G-2 bits. 

Additionally, the spec indicates that "When G ≥ 2 and pmpcfgi.A[1] is set, i.e. the mode is NAPOT, then bits pmpaddri[G-2:0] read as all ones." I don't see this being implemented correctly either.

Could you comment on these two issues, please?

aswaterman, 2021-04-27:
You sure about that?  The pmpaddr read logic in `processor_t::get_csr()` appears to be handling this correctly (by relying on `pmp_tor_mask()`, which is a function of G).  Similarly, the PMP-checking logic in `mmu_t::pmp_ok()` depends on `pmp_tor_mask()`.

aprnath, 2021-04-27:
No, I am not sure about that.. :-)
I am assuming that `PMP_SHIFT` is used to specify granularity, right?

Then:
```
state.pmpaddr[i] = val & ((reg_t(1) << (MAX_PADDR_BITS - PMP_SHIFT)) - 1);
```
seems to imply that `state.pmpaddr[i]` has `PMP_SHIFT` less bits..

Thats what I see when I build with `PMP_SHIFT=6` (64byte granularity) - i.e a nibble extra seems to be eaten up. I have a 40 bit address (`MAX_PADDR_BITS=40`), and `PMP_SHIFT=6`). Then writing all 1s to `pmpaddr0` and corresponding `pmpcfg0` index, I get
```
    3 0x000000008000003c (0x00736333) x 6 0x00ffffffffffffff    
    core   0: 0x0000000080000040 (0x3b031073) csrw    pmpaddr0, t1    
    3 0x0000000080000040 (0x3b031073)  csr 0x000003b0 0x00000003ffffffff   

    core   0: 0x000000008000004c (0x3a031073) csrw    pmpcfg0, t1
    3 0x000000008000004c (0x3a031073)  csr 0x000003a0 0x000000000000009f
    core   0: 0x0000000080000050 (0x3b002373) csrr    t1, pmpaddr0
    3 0x0000000080000050 (0x3b002373) x 6 0x00000003ffffffff
```
I could be barking up the wrong tree entirely, of course.

aswaterman, 2021-04-27:
Ah.  `PMP_SHIFT` isn't meant to be changed by users: it must always be 2, irrespective of G.

The mechanism for changing `G` is to call `set_pmp_granularity`.  Normally this is done by passing a DeviceTree to Spike that includes the `riscv,pmpgranularity` property.  But you could also manually hack Spike to call `set_pmp_granularity(64)`.

aprnath, 2021-04-27:
Thanks Andrew. That helps. 

Where do you recommend I add this call?

aswaterman, 2021-04-27:
Probably nearby where the current call is in sim.cc?

aprnath, 2021-04-27:
Thanks for the assist, Andrew. This
```
  if (n_pmp > 0) {
    // For backwards compatibility with software that is unaware of PMP,
    // initialize PMP to permit unprivileged access to all of memory.
    set_csr(CSR_PMPADDR0, ~reg_t(0));
    set_csr(CSR_PMPCFG0, PMP_R | PMP_W | PMP_X | PMP_NAPOT);
  }
```
threw me off a bit, as I was expecting the `pmpcfg*` to be cleared out of reset.

Thanks!

closed 2021-04-27 by aprnath

aswaterman, 2021-04-28:
👍

talhashahzad12345, 2024-06-04:
Hi,
Is there a cli based way to clear out pmpcfg* at reset instead of starting off with PMP_NAPOT?

aswaterman, 2024-06-04:
I recommend doing this in the software running on the machine.  For machines that reset the PMPs to OFF, this would just be a few extra instructions, but otherwise harmless.