riscv-software-src/riscv-isa-sim #704: Can Spike support rocc instrcution multi issue?

xiongwu12345, 2021-04-27:
I know in spike, there is a rocc port to support riscv custom extension instructions.But if this rocc port can support issue several extension instructions at a time?If can, how to program this function?
Could someone offer me some help?Any information is appreciated.Thanks a lot!

aswaterman, 2021-04-28:
Spike does not model any particular microarchitecture.  Instructions are always executed one at a time.

xiongwu12345, 2021-05-06:
Oh,thank you.But I wonder if I write a particular microarchitecture that support muti-issue and it can support riscv custom extension instructions through a rocc port, so can I can find some methods to get a certain amount of custom instructions?