riscv-software-src/riscv-isa-sim #707: Incorrect VSSI interrupt handling when going from HS-mode to VS-mode

vipulgupta-sifive, 2021-05-06:
I am trying new hypervisor support in Spike and getting the incorrect behavior as per my understanding.
I am trying a C test to trigger VSSI interrupt from VS-mode by setting HIP.VSSIP==1 and SSTATUS.SIE==0.

As per my understanding, VSSI trap should be taken when we go from HS-mode to VS-mode in this case regardless of SSTATIS.SIE

I have attached the disassemble code, program.elf and spike command line to this and Following is the snippet of my C-code
after being reset to M-mode and hypervisor is enabled:
    goto_priv(PRIV_HS);
    CSRC(sstatus, SSTATUS_SPIE_BIT | SSTATUS_SIE_BIT);
    CSRS(CSR_HIE, 0x4);
    CSRS(CSR_HIP, 0x4);
    CSRS(CSR_HVIP, 0x4);
    goto_priv(PRIV_VS);
    
I am expecting that I after the last statement we will take VSSI trap into HS-mode but that did not happen

Please note that if I change the above code like the following:
    goto_priv(PRIV_HS);
    CSRC(sstatus, SSTATUS_SPIE_BIT | SSTATUS_SIE_BIT);
    CSRS(CSR_HIE, 0x4);
    CSRS(CSR_HIP, 0x4);
    CSRS(CSR_HVIP, 0x4);
    **CSRS(sstatus, 0x2);  //ENABLE SSTATUS.SIE**
    goto_priv(PRIV_VS);

Then spike takes a trap to HS mode.

Let me know if you need more data from me on this.

[asm.txt](https://github.com/riscv/riscv-isa-sim/files/6437990/asm.txt)

[spike.sim.rerun.txt](https://github.com/riscv/riscv-isa-sim/files/6438007/spike.sim.rerun.txt)
[program.elf.txt](https://github.com/riscv/riscv-isa-sim/files/6438008/program.elf.txt)

aswaterman, 2021-05-06:
@avpatel I haven't done my part to triage this yet, but I'll take a look sometime in the next couple days.  LMK if you get to it first.

aswaterman, 2021-05-09:
It looks like this test _passes_.  Did you submit the version that fails or your modified version?

Some general advice on reporting failing test cases:
- Try minimize the test case to the bare minimum necessary to manifest the bug.  This is a lengthy test (> 20 KiB of code).
- It looks like this test relies on vendor-specific features, so it cannot execute correctly on standard Spike.  Try out the test case on standard Spike before submitting.

aswaterman, 2021-05-09:
Anyway, not sure what I can do to help here, since everything seems to be working properly.

Prior to the SRET, I see that `sstatus.SIE`=0, and `vsstatus.SIE`=1.  After the SRET, I see `sstatus.SIE`=1, and then an interrupt is taken.  This all seems like correct behavior to me.

avpatel, 2021-05-09:
There is a bug in your test. You need to set HIDELEG.VSSIP to allow VS-MODE take SSIP interrupt.

Please try that and let us know.

@aswaterman, sorry for delayed response. I was busy with AIA support in QEMU, OpenSBI, and Linux.

Regards,
Anup

aswaterman, 2021-05-09:
@avpatel No problem, 2-day response time including a weekend day is pretty good IMO :)

vipulgupta-sifive, 2021-05-09:
@avpatel the intention of the test is to take exception in HS-mode and not VS-mode. That is why HIDELEG.VSSIP is not set.

vipulgupta-sifive, 2021-05-09:
@aswaterman I will take a look at the test again and try to get a smaller version. This test is part of the bigger infrastructure that is why you get such a long trace. It is going to be some work to reproduce on the smaller test. Will try and get back

vipulgupta-sifive, 2021-05-09:
@avpatel. Also my understanding is that higher level global exception is always enabled regardless of SIE bit. So in VS-mode, HS-mode global interrupts will always be enabled and since this interrupt is delegated to HS-mode and NOT vs-mode exception should be taken in HS-mode.

Is this understanding incorrect?

aswaterman, 2021-05-10:
@vipulgupta-sifive Once again, the test you submitted _passes_.  We do not observe a failure when executing it.

vipulgupta-sifive, 2021-05-10:
Andrew, original test was doing multiple scenarios and reported failure only on one scenario. I was experimented with few things and it is possible that correct return status is not captured.

I am going to work on this test and make sure it fails as I mentioned. I will get back tomorrow.

Regards
Vipul

vipulgupta-sifive, 2021-05-10:
@aswaterman , I have made the test changes to make sure that this is the only scenario in the test and also removed the debug statements to reduce the code. I see the test fail in my area. I am using Spike version "Spike RISC-V ISA Simulator 1.0.1-dev " which is the same as public version I believe.

Attached are the new asm.txt, program.elf, spike.sim.rerun and spike.log from my run

Let me know if you need anything else.
[asm.txt](https://github.com/riscv/riscv-isa-sim/files/6452988/asm.txt)
[spike.sim.rerun.txt](https://github.com/riscv/riscv-isa-sim/files/6452991/spike.sim.rerun.txt)
[spike.log.txt](https://github.com/riscv/riscv-isa-sim/files/6452993/spike.log.txt)
[program.elf.txt](https://github.com/riscv/riscv-isa-sim/files/6452995/program.elf.txt)

closed 2021-05-11 by vipulgupta-sifive