riscv-software-src/riscv-isa-sim #709: Spike not raising LGPF exception when HLVX.WU tries to acces non-exec Guest page in HS-mode

vipulgupta-sifive, 2021-05-17:
@avpatel 

This test is trying to access Guest page with non-exec bit clear using HLVX.WU instruction in HS-mode. As per spec Guest Page Fault should be raise here but spike doesn't raise any exception.

Please note that I I print the page walk for the VA accessed and it looks like the following showing that PTE indeed had X==0.

stage=1 translation for va=0x100001000
    s/hg/vs/atp=0x8000000000080018, mode=0x8, ppn=0x80018000
        LVL2: PTE@0x80018020, pte_val=0x20006801
            Non-Leaf pte_ppn=0x8001a000
        LVL1: PTE@0x8001a000, pte_val=0x20006c01
            Non-Leaf pte_ppn=0x8001b000
        LVL0: PTE@0x8001b008, pte_val=0x400004cf
            Leaf pte_ppn=0x100001
                R
                W
                E
                A
                D
GVA=0x100001000
stage=2 translation for va=0x100001000
    s/hg/vs/atp=0x8000000000080014, mode=0x8, ppn=0x80014000
        LVL2: PTE@0x80014020, pte_val=0x20003401
            Non-Leaf pte_ppn=0x8000d000
        LVL1: PTE@0x8000d000, pte_val=0x20003801
            Non-Leaf pte_ppn=0x8000e000
        LVL0: PTE@0x8000e008, pte_val=0x240004d7
            Leaf pte_ppn=0x90001
                R
                W
                U
                A
                D

Attached is disassembly, elf file and spike command line that I used. 
[asm_lgpf.txt](https://github.com/riscv/riscv-isa-sim/files/6497057/asm_lgpf.txt)
[program.elf.lgpf.txt](https://github.com/riscv/riscv-isa-sim/files/6497059/program.elf.lgpf.txt)
[spike.sim.rerun.lgpf.txt](https://github.com/riscv/riscv-isa-sim/files/6497062/spike.sim.rerun.lgpf.txt)

Please note that I also have a test failing when Stage 1 page walk lands up on the non-exec leaf PTE using the HLVX.WU and in this case expectation is to have Load Page Fault exception but spike doesn;t raise that either. Let me know if you need that test separately or if you can fix both LGPF and LPF issue with this test.

Thanks
Vipul

avpatel, 2021-05-23:
I am still busy with other things. I try this soon at my end. If you already have a root caused it then let me know.

Regards,
Anup

vipulgupta-sifive, 2021-05-24:
No I haven't root caused it

jklockars, 2021-05-27:
We ran into this issue recently as well, but have not gotten around to reporting it.
Spike appears to be handling the HLVX as if it was a read with MXR set. Unlike that case, however, the specification says that R is not enough here, X is required.

avpatel, 2021-05-27:
Please replace "mxr = true" with "type = FETCH" in mmu::translate() function of riscv/mmu.c source file line #69.

If this works then I send a fix PR.

Regards,
Anup

jklockars, 2021-05-27:
That won't quite do it, since the PMP check must still be done for R rather than X.

scottj97, 2021-09-05:
I believe #751 fixed this?

scottj97, 2021-12-07:
Closing, as I believe this is now fixed, and OP hasn't responded.

closed 2021-12-07 by scottj97