riscv-software-src/riscv-isa-sim #710: How could we get the value of the misa register?

zhengxiaolinX, 2021-05-19:
Hi,

I've tried 
```
int misa;
asm volatile("csrr x0, misa" : "=r" (misa));
printf("%x\n", misa);
```
and got an `An illegal instruction was executed!` message in spike. I wonder how can I get the value of this register, or how could we get CPU features of the current simulator, e.g. whether C extension supported. I am not so familar with this, and thanks in advance.

aswaterman, 2021-05-19:
`misa` is only readable in M-mode.  If you are running a user program using `pk`, your program is running in U-mode, it is impossible to directly read that register.

zhengxiaolinX, 2021-05-19:
> `misa` is only readable in M-mode. If you are running a user program using `pk`, your program is running in U-mode, so it is impossible to directly read that register.

Thank you @aswaterman for replying to this! 
Just one more question, I wonder how to detect whether the C extension is supported on the current core by using some instructions. Thanks!

aswaterman, 2021-05-19:
There isn't a straightforward way to check.  You can try executing a C-extension instruction and see if you get an exception.

On RISC-V Linux, a program's ELF header indicates whether it requires the C extension, and the kernel's ELF loader will refuse to load the program if the processor does not support C.

zhengxiaolinX, 2021-05-19:
> There isn't a straightforward way to check. You can try executing a C-extension instruction and see if you get an exception.
>
> On RISC-V Linux, a program's ELF header indicates whether it requires [...]

Thank you, I really appreciate your help and the details.

closed 2021-05-19 by zhengxiaolinX