riscv-software-src/riscv-isa-sim #711: How to execute c programs in spike using extensions such as M,A,F,B,V and others.

nikita890-art, 2021-05-21:
I am currently using a 32-bit spike and i am trying to run my c code with extensions such as M,A,F,B,V etc. , but i dont really know on what command to give in the terminal so as to execute the program with these extensions.
Presently i am using these commands for compiling and to run,
riscv32-unknown-elf-gcc <file name>
RISCV/riscv-isa-sim/build/spike  --isa=RV32IMAC   RISCV/riscv-pk/build/pk  a.out

What changes do i need to make in these commands to execute it using extensions?

LakshmiKanth-77, 2021-05-21:
You should compile it with same extension and then simulate with same extension .
riscv32-unknown-elf-gcc -march= rc32imaf .c file -o object file[some letter]

nikita890-art, 2021-05-28:
thank you!!!!it works now

closed 2021-05-28 by nikita890-art