riscv-software-src/riscv-isa-sim #715: What is the use of compression extension "C" in the RISC-V ISA?

nikita890-art, 2021-05-26:
I have executed a c program with and without the C extension and I get different amount of lines for the assembly code for both executions. What is the C extension doing to the c code exactly?

mwachs5, 2021-05-26:
It is taking the assembly code that comes out of the compiler and turning some common 32-bit instructions into their 16-bit shorthand equivalents ("Compressing" them).

nikita890-art, 2021-05-26:
Thank you so much..
I get more number of instructions and clocks when I use the C extension compared to when I dont use it.
Isnt C extension suppose to reduce the number of instructions and clock?

mwachs5, 2021-05-26:
no. C extension only reduces the size of the binary, it doesn't say anything about how long it takes to execute the instructions

nikita890-art, 2021-05-28:
oh..ok...thank you!!