riscv-software-src/riscv-isa-sim #720: LR/SC sequence incorrect execution

erito27, 2021-06-01:
Hello all,
In section 8.2 of the RISCV Unprivileged spec it states that

> The SC must fail if a store from another hart to the address range reserved by the LR can be observed to occur between the LR and the SC.

I've created a simple case where I create a reservation, followed by a regular store to the same address and I'm seeing the following SC is succeeding.

```
core   0: 0x0000000000010124 (0x10053c2f) lr.d    s8, (a0)
core   0: 0x0000000000010128 (0x00f53023) sd      a5, 0(a0)
core   0: 0x000000000001012c (0x19053aaf) sc.d    s5, a6, (a0)
: reg 0 s5
0x0000000000000000
```

I was expecting the SC.D to fail after. 

Thanks,
elliott

aswaterman, 2021-06-01:
The operative word is "store from another hart".  This is a store from the same hart.

erito27, 2021-06-01:
I'm somewhat new to RISCV and I might be a bit confused about what a hart is. Basically this is saying that the SC will fail if a different core writes somewhere within the reservation?

Thanks for your help!

aswaterman, 2021-06-01:
Ah, yes... "hart" = "hardware thread", so what you wrote is basically right.

scottj97, 2021-09-05:
@erito27 please close this issue if you are satisfied with its resolution.

closed 2021-09-07 by erito27