riscv-software-src/riscv-isa-sim #731: Issue when running SPEC CPU 2017 applications

sawansib, 2021-06-27:
I'm using Spike as a frontend for our in-house simulator. Several applications from SPEC CPU are failed to execute when running with Spike pk. For example, the application 619.lbm_s failed with the message ```LBM_allocateGrid: could not allocate 1611.3 MByte``` even if I use spike -m20480. Application 620.omnetpp_s fails to find the input file but works fine when running without spike directly on x86 machine. The exact command line i used for both the applications are below:-
```spike -m20480 --trace-location /home/ssingh/riscv-traces --trace-name 619.lbm_s pk ./lbm_s_base.riscv64-m64 2000 reference.dat 0 0 200_200_260_ldc.of```
```spike --trace-location /home/ssingh/riscv-traces --trace-name 620.omnetpp_s pk ./omnetpp_s_base.riscv64-m64 -c General -r 0```

Sawan

aswaterman, 2021-06-27:
Older versions of PK support only approx. 2 GB of memory, no matter how much memory the system actually has.  The limit has been lifted with newer versions of PK.

(Note also that some versions of the Newlib C library might have similar limitations; you'll need to [statically] link against glibc, i.e., use the riscv64-unknown-linux-gnu toolchain, not riscv64-unknown-elf.)

sawansib, 2021-06-28:
Hi Andrew!

Thank you for the response you were right. The memory issue is solved by updating the pk.

Sawan

closed 2021-06-28 by sawansib