riscv-software-src/riscv-isa-sim #734: How to read/write all hart's registers & memory

chenc6, 2021-06-29:
Hi folks, 

I am recently trying to understand the functionalities of the debug module in Spike.

I check the RISC-V Debug Support Version 1.0.0-STABLE from this repository [https://github.com/riscv/riscv-debug-spec](url)

I would like to know whether Spike contains the following two features. If so, how should I use them?:

1. All hart registers (including CSRs) can be read/written. 
    I think Spike has an Interactive Debug Mode that allows users to READ contents of specific registers and memory locations. Can we WRITE data to them before executing each instruction in debug mode?
2. Arbitrary instructions can be executed on a halted hart.
    My understand is that the Program Buffer is developed to handle this function and allow users to run additional instructions on a halted hart. In riscv/debug_module.h, it seems like Spike implemented this feature. How should I utilize this feature?

The above two features are from RISC-V Debug Support Version 1.0.0-STABLE, Chapter 1.5, 1&2 and 12 respectively.

I am new to this field. Any helps would be much appreciated!

timsifive, 2021-06-29:
Spike contains 2 debug methods:

1. The interactive method, which you enter when hitting ^C. This has nothing to do with the debug spec.
2. It implements the debug spec, version 1, with support for accessing registers using abstract commands, as well as through the program buffer. You can use OpenOCD to connect to spike, and gdb in turn can connect to OpenOCD.

What are you trying to achieve?

chenc6, 2021-06-30:
Hi @timsifive 

Thank you for your help! I may misunderstand the interactive method and debug specification.
I want to modify the contents of registers and memory before executing each instruction from a program.

Here is an example that I want to achieve, please check:
`r1=0,r2=1,r3=1`
`Data store in memory ADDR is 2`
Main program:
`L1: ADD r1,r2,r3; \\ r1 = r2+r3 = 2`
`L2: Li r2, ADDR; \\ r2 = 2`

Let's say I set a breakpoint at the beginning of this program and want to execute instructions (L1 & L2) one by one in debug mode. Before executing L1, I want to modify the value stored in a register (r3) from "1" to "3". So after that, the value in r1 will be 4. Also, I want to modify the value stored in the memory (ADDR) from "2" to "4". So, the value in r2 will also be 4.

My understand is that:
1. If all hart registers (including CSRs) can be read/written, then there may be some commands I can directly use to update value in registers & memory.
2. If no such functions/commands, I can put some additional instructions into the program buffer and run them before I execute L1 and L2. 

I hope this is clear.

timsifive, 2021-06-30:
The program buffer is for debuggers to use. OpenOCD writes to the program buffer (if necessary) to do what gdb asks it to do. Generally you'd use gdb-OpenOCD-spike and then you can do whatever you can usually do with gdb. The README describes how to run a simple program using gdb.

I don't think spike's interactive debug mode supports setting breakpoints.

chenc6, 2021-07-01:
Hi @timsifive 

Thank you for your help! Your explanation solves lots of my confusion of debug mode. 

I will try to use gdb to setup breakpoints and learn how to utilize abstract commands to achieve my purpose. 

I will close this issue and would like to ask you for help in future if necessary.

closed 2021-07-01 by chenc6