riscv-software-src/riscv-isa-sim #741: Spike Vs dromajo trace mismatch

sammy17, 2021-07-09:
I ran a RISCV-DV generated testcase with both Spike and dromajo and got an instruction trace mismatch. 

Mismatch occurs right after the test jumps to the user mode privilege (with an MRET after initializing user mode). 

dromajo throws an exception on the first user-mode instruction while spike proceeds executing the user-mode program without any issue.  

I am running a Rocket core simulation as well and the Rocket core simulation trace is similar to the dromajo trace. 
I could not think of any obvious reason for this mismatch, really appreciate any help to identify the reason for this. 

I have attached the assembly testcase, spike trace, dromajo trace and disassembly of the testcase. 

**Command used to generate Spike trace:**
`spike --log-commits --isa=RV64G test.elf`

**Spike Vs dromajo diff:**
[image]
[test.zip](https://github.com/riscv/riscv-isa-sim/files/6788220/test.zip)

aswaterman, 2021-07-09:
My best guess is that the issue relates to the initial state of the PMP registers, which is mostly unspecified.  If I'm right, then both simulators are potentially correct: the bug is that the software test is relying on unspecified behavior, namely the reset value of various CSRs.

sammy17, 2021-07-09:
You are correct. Once I initialize the PMP registers at the start of the program, both Spike and dromajo gave the same trace. Thank you very much!

sammy17, 2021-07-09:
Closing this issue since it's resolved once I initialize the PMP registers in the program.

closed 2021-07-09 by sammy17