riscv-software-src/riscv-isa-sim #745: sc.w fails in spike even though the reservation is still valid

d-thakkar, 2021-07-13:
I am encountering a comparison mismatch with my core and spike for lr.w and sc.w instructions. This does not happen for all lr and sc combinations but for some of them only.

In the following snippet, for PC 8001ae0c, spike updates rd to 00000001(non zero) value flagging that sc.w instruction to be failed. However there was no load or update to the reserved memory location of 0x8003b06c after lr.w instruction at PC 8001ade0 keeping that address reserved.  Since there was no load instruction after the lr.w instruction, why should the following sc.w instruction be flagged as failed?

[image]

aswaterman, 2021-07-13:
SC is permitted to fail for any reason, as long as an LR/SC loop that adheres to the forward-progress constraints eventually succeeds.

In this case, I believe the issue is that there is a CSR access instruction between the LR and SC. That violates one of the forward-progress constraints, and so the SC is permitted to fail forever.

Spike will also occasionally cause SCs to fail if Spike decides to context switch between the harts it is simulating. (This will happen even when simulating only a single hart.) But as long as the LR/SC loop adheres to the forward-progress constraints, the SC will eventually succeed.

In short, this seems like legal behavior.

d-thakkar, 2021-07-14:
Hey Thanks for helping. That helps understand the failure. But in my test I also have other LR/SC loop that has CSR access instructions between LR and SC, but they are not flagged. 

From spec "LR/SC sequences that do not lie within constrained LR/SC loops are unconstrained. Unconstrained
LR/SC sequences might succeed on some attempts on some implementations, but might never
succeed on other implementations."

If that's the case, how do you attain parity in terms of validation ?

aswaterman, 2021-07-15:
I might’ve been wrong about exactly why Spike chose to fail the SC in this case, but of course the broader point remains that Spike is not doing anything incorrect.

Usually the strategy for using spike as a cosimulator is to allow the DUT to override spike for things like SC success/failure. These overrides are necessary for other things, anyway, like interrupt timing or data sharing in multiprocessors.

d-thakkar, 2021-07-23:
Thanks for the help @aswaterman. I was able to get past the failure. Closing this issue.

closed 2021-07-23 by d-thakkar